The-OpenROAD-Project/OpenLane #1159: Use .odb rather than .def between OR steps

maliberty, 2022-06-22:
### Prompt
Between steps using openroad the intermediate files should be a .odb rather than a .def.  We are storing additional information that can't be represented in def (guides, hierarchy, etc).  ORFS has already made the switch as a reference.

### Proposal
Use read_db / write_db rather than read_def / write_def

donn, 2022-06-23:
This has been proposed before. I'd like to do it, but then I'd have to account for "contamination" for if the DEFs are manipulated outside of OpenROAD. But it would be good not just for the additional information, but also just for performance reasons.

maliberty, 2022-06-23:
If required you could always write_def and read_def for outside manipulation.  Better would be to rewrite such as an openroad tcl using the odb apis instead.

Should I take your comment as a rejection or just a concern?

donn, 2022-06-23:
Concern. It's definitely planned. Just that it's going to be quite a lift.

closed 2022-09-19 by donn