The-OpenROAD-Project/OpenLane #1161: Pull openlane and make mount use different docker images

armleo, 2022-06-24:
### Description
Pull openlane and make mount use different docker images

https://github.com/The-OpenROAD-Project/OpenLane/blob/a633b1fd2529851b284d7704c7f45cad5222f031/Makefile#L94
https://github.com/The-OpenROAD-Project/OpenLane/blob/a633b1fd2529851b284d7704c7f45cad5222f031/Makefile#L102

When running with 
```
export OPENLANE_IMAGE_NAME=efabless/openlane:ebad315d1def25d9d253eb2ec1c56d7b4e59d7ca-amd64
make mount
```
I get an error, despite this working fine in the past. Understandable that the new flow wanted to have amd64/arm in separate variable but then why does make pull-openlane use the old syntax?

armleo, 2022-06-24:
After the pull request is approved, leave this open so I can update the documentation on how to properly set the env variable `OPENLANE_IMAGE_NAME` .

donn, 2022-06-24:
OpenLane makes use of Docker manifests, in which `docker pull` and `docker mount` will automatically pull the right version for your architecture. The only reason `DOCKER_ARCH` exists in the mount command is to be able to run amd64 OpenLane on arm64v8 and vice versa. What likely happen is that you `git pull`ed in the tiiiiny window between the x86 image being pushed and the arm64v8 image being pushed. This is why by default `make openlane` will attempt to build it if pulling doesn't work first try.

armleo, 2022-06-24:
"will automatically pull the right version"

That is correct. Unless I overwrite the variable in that case the `make mount` that builds PDK and related will error out complaining with following: 

```
the manifest with the name : 	docker run --rm -v /home/armleo/Desktop/OpenLaneGSOC:/openlane -v /home/armleo/Desktop/OpenLaneGSOC/designs:/openlane/install -v /home/armleo/Desktop/OpenLaneGSOC/pdks:/home/armleo/Desktop/OpenLaneGSOC/pdks -e PDK_ROOT=/home/armleo/Desktop/OpenLaneGSOC/pdks -e STD_CELL_LIBRARY=sky130_fd_sc_hd --user 1000:1001 -e DISPLAY=:0.0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/armleo/.Xauthority:/.Xauthority --network host efabless/openlane:5a767e7fd208359f89d6784c8bc7533843013962-amd64- sh -c "./flow.tcl -design spm -tag openlane_test -overwrite"
Unable to find image 'efabless/openlane:5a767e7fd208359f89d6784c8bc7533843013962-amd64-' locally
docker: Error response from daemon: manifest for efabless/openlane:5a767e7fd208359f89d6784c8bc7533843013962-amd64- not found: manifest unknown: manifest unknown.
```

If I overwrite without -amd64 part it will error out in building PDK stage complaining about the docker not existing.

Anyway, the pull request should fix this.

donn, 2022-06-24:
Let me test that and get back to you.

armleo, 2022-06-24:
Hmm... Seems something weird is happening. The issue is no longer reproducible. But maybe it's worth merging the pull request above anyway, so there are no weird behaviors.

I am able to reproduce a different issue. When I set the `OPENLANE_IMAGE_NAME` to the variant without -amd64 the `pull-openlane` works fine, but the `make mount` fails complaining about efabless/openlane:5a767e7fd208359f89d6784c8bc7533843013962-amd64-amd64 not existing.

donn, 2022-07-18:
Fixed in #1205.

closed 2022-07-18 by donn