The-OpenROAD-Project/OpenLane #1165: [Error PDN-0179] Unable to repair all channels

AyaseErii, 2022-06-25:
### Description
Hello, I am trying to use OpenLane to place macros for a core design, and I set `DESIGN_IS_CORE` as `1` and `FP_PDN_CORE_RING` as `1`. And I want to make sure the automatic marco placement on the core design can work at first. However, when the process went into the stage of PDN, it shows the error (see Logs). Could anyone help me out with what is wrong here? I will appreciate it!

### Environment
[image]

### Reproduction Material
* set `DESIGN_IS_CORE` as `1` and `FP_PDN_CORE_RING` as `1` in the `config.tcl`
* add necessary commands referred from the chip integration tutorial

### Expected behavior
Auto macro placement should work successfully.

### Logs
```
    OpenROAD 0b8b7ae255f8fbbbefa57d443949b84e73eed757 
    This program is licensed under the BSD-3 license. See the LICENSE file for details.
    Components of this program may be licensed under more restrictive licenses which must be honored.
    [INFO ODB-0222] Reading LEF file: /openlane/designs/mydesign_core/runs/RUN_2022.06.24_20.24.07/tmp/merged.unpadded.nom.lef
    [INFO ODB-0223]     Created 13 technology layers
    [INFO ODB-0224]     Created 25 technology vias
    [INFO ODB-0225]     Created 458 library cells
    [INFO ODB-0226] Finished LEF file:  /openlane/designs/mydesign_core/runs/RUN_2022.06.24_20.24.07/tmp/merged.unpadded.nom.lef
    [INFO ODB-0127] Reading DEF file: /openlane/designs/mydesign_core/runs/RUN_2022.06.24_20.24.07/results/floorplan/mydesign_core.def
    [INFO ODB-0128] Design: mydesign_core
    [INFO ODB-0130]     Created 296 pins.
    [INFO ODB-0131]     Created 74280 components and 206305 component-terminals.
    [INFO ODB-0133]     Created 17456 nets and 34951 connections.
    [INFO ODB-0134] Finished DEF file: /openlane/designs/mydesign_core/runs/RUN_2022.06.24_20.24.07/results/floorplan/mydesign_core.def
    [INFO PDN-0001] Inserting grid: stdcell_grid
    [WARNING PDN-0178] Remaining channel (525.7800, 2896.5600) - (609.0400, 2943.2800) on met1 for nets: VPWR, VGND
    [WARNING PDN-0178] Remaining channel (554.7600, 2855.7600) - (609.0400, 2896.5600) on met1 for nets: VPWR, VGND
    [WARNING PDN-0178] Remaining channel (828.4600, 2654.4800) - (955.8800, 2665.8400) on met1 for nets: VGND
    [WARNING PDN-0178] Remaining channel (837.2000, 2170.3200) - (970.1400, 2181.6800) on met1 for nets: VGND
    [WARNING PDN-0178] Remaining channel (1153.2200, 3179.4400) - (1305.0200, 3193.5200) on met1 for nets: VPWR, VGND
    [WARNING PDN-0178] Remaining channel (1232.3400, 2888.4000) - (1296.2800, 2926.9600) on met1 for nets: VGND
    [WARNING PDN-0178] Remaining channel (1557.1000, 1781.3600) - (1678.0800, 1792.7200) on met1 for nets: VPWR
    [WARNING PDN-0178] Remaining channel (1868.0600, 2200.2400) - (2018.4800, 2217.0400) on met1 for nets: VPWR
    [WARNING PDN-0178] Remaining channel (1983.9850, 3013.5200) - (1986.2800, 3139.1200) on met1 for nets: VPWR, VGND
    [WARNING PDN-0178] Remaining channel (405.0400, 2861.3960) - (1021.0400, 2952.9130) on met4 for nets: VGND
    [WARNING PDN-0178] Remaining channel (405.0400, 2861.3960) - (1328.2400, 2958.9000) on met4 for nets: VGND
    [WARNING PDN-0178] Remaining channel (712.2400, 2925.4250) - (1328.2400, 2958.9000) on met4 for nets: VGND
    [ERROR PDN-0179] Unable to repair all channels.
    PDN-0179
```

Thanks!

maliberty, 2022-06-25:
Please attach a test case

AyaseErii, 2022-06-25:
Thanks for your reply, Matt! Here you go:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8985475/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-06-27:
@AyaseErii 
Can you share your `/openlane/designs/mydesign_core/runs/RUN_2022.06.24_20.24.07/config.tcl` ?

vijayank88, 2022-06-27:
@AyaseErii 
If you using OpenLane flow by default macro halo spacing set to 0.
```
set ::env(PL_MACRO_CHANNEL) {0 0}
set ::env(PL_MACRO_HALO) {0 0}
```
So update with min spacing value like 5 or 10 and run the flow again.

AyaseErii, 2022-06-27:
@arlpetergadfort Thanks for your quick check! I am not sure why there are so many overlaps happening among macros. I believe those macros are separately generated and some of them should be placed repeatedly according to the core design.

AyaseErii, 2022-06-28:
@vijayank88 Thanks for the suggestion! I set the spacing as below:
```
set ::env(PL_MACRO_CHANNEL) {40 40}
set ::env(PL_MACRO_HALO) {40 40}
```
This did solve the overlapping issue! However, the tool shows another error in the stage 17 resizing:
```
[STEP 17]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/mydesign_core/runs/RUN_2022.06.27_15.34.28/logs/routing/17-resizer.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 85370
[INFO GRT-0198] Via related Steiner nodes: 183
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 447743
[INFO GRT-0112] Final usage 3D: 7584702
[ERROR GRT-0118] Routing congestion too high.
Error: resizer_routing_timing.tcl, 53 GRT-0118
child process exited abnormally
```
Again, I put the newly repo file with config.tcl inside as below:
[openroad_issue_repo_with_config_after_setting_spaces.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8997213/openroad_issue_repo_with_config_after_setting_spaces.tar.gz)

Any suggestion regarding this error? I will appreciate it!

Thanks!

AyaseErii, 2022-06-29:
@vijayank88 Thanks again for the suggestion. It looks like `set ::env(DIODE_INSERTION_STRATEGY) 4` does not help with passing stage 17 resizer. Pleased find the repo with config.tcl below:
[openroad_issue_reproducible_with_config_tcl.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9011120/openroad_issue_reproducible_with_config_tcl.tar.gz)

It shows the error message that the routing congestion too high. I am not sure what that means.
```
[STEP 17]
[INFO]: Running Global Routing Resizer Timing Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/fpga_core/runs/RUN_2022.06.29_02.37.37/logs/routing/17-resizer.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 85370
[INFO GRT-0198] Via related Steiner nodes: 183
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 447743
[INFO GRT-0112] Final usage 3D: 7584702
[ERROR GRT-0118] Routing congestion too high.
Error: resizer_routing_timing.tcl, 53 GRT-0118
child process exited abnormally
```

BTW, do you think manual macro placement will be helpful?

vijayank88, 2022-06-30:
Yes you can try manual macro placement as well.

But it will be good if you provide git repo link with all input data, to debug or try with other configuration for auto macro placement.

vijayank88, 2022-06-30:
@AyaseErii 
Check below layout view. There is wrong placement, some connection going outside the DIE area.
[image]

vijayank88, 2022-07-01:
@AyaseErii 
have you tried manual macro placement?

vijayank88, 2022-07-01:
I cloned your source files and running it. More than 3 hrs still macro placement going on. Will update once flow completes.

vijayank88, 2022-07-02:
@AyaseErii 
I just took relative die size calculated by tool. DIE_AREA is ` 0 0 11695 11705`. Now running global routing. 
Is that die size ok?

AyaseErii, 2022-07-02:
> @AyaseErii I just took relative die size calculated by tool. DIE_AREA is ` 0 0 11695 11705`. Now running global routing. Is that die size ok? But still seeing below warning:
> 
> ```
> [WARNING GRT-0040] Net VGND has wires outside die area.
> [WARNING GRT-0040] Net VPWR has wires outside die area.
> [WARNING GRT-0040] Net VPWR has wires outside die area.
> ```

@vijayank88 Yes, the die size is ok. Just out of curiosity, how do you let the tool calculate the relative die size? Do you use `::env(FP_CORE_UTIL) "10"` or something else?

I am not sure if the warning will really make the flow fail.

AyaseErii, 2022-07-06:
@vijayank88 Hi, how is the flow going? Could you let me know if you successfully got the GDSII? Thanks!

vijayank88, 2022-07-07:
@AyaseErii 
Update OpenROAD app: https://github.com/The-OpenROAD-Project/OpenLane/blob/b718fd064063333703734ce3ae8bb56aa6810b0b/dependencies/tool_metadata.yml#L61 to latest commit: 809c4384658c8b4e9f37b47252ff2a23c4f866c1 and run the flow. It will pass global routing congestion issue.

kareefardi, 2024-07-17:
Please open a new issue if you think there is still a problem. The latest comment indicate that this has been solved.

closed 2024-07-17 by kareefardi