The-OpenROAD-Project/OpenLane #1170: OpenLane Yosys Error during synthesize

akifcelik, 2022-06-28:
### Description
Hi everyone, I am trying to use openlane and skywater pdk to synthesize my ALU  design . When start the flow my config.tcl file i get the following error:

[INFO]: Running Synthesis...
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/ALU/runs/RUN_2022.06.28_18.49.22/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:

49. Executing ABC pass (technology mapping using ABC).

49.1. Extracting gate netlist of module `\divider_64bit_test' to `/tmp/yosys-abc-NTqtCv/input.blif'..
Extracted 24759 gates and 24888 wires to a netlist network with 129 inputs and 256 outputs.

49.1.1. Executing ABC.
sh: line 1:   140 Aborted                 /build/bin/yosys-abc -s -f /tmp/yosys-abc-NTqtCv/abc.script 2>&1
ERROR: ABC: execution of command "/build/bin/yosys-abc -s -f /tmp/yosys-abc-NTqtCv/abc.script 2>&1" failed: return code 134.
child process exited abnormally

[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/ALU/runs/RUN_2022.06.28_18.49.22/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/ALU/runs/RUN_2022.06.28_18.49.22/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(SYNTH_BIN)  -c $::env(SYNTH_SCRIPT)  -l [index_file $::env(synthesis_logs)/synthesis.log]  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 32)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 55)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        if { [info exists arg_values(-file)] } {
                run_file [file normalize $a..."
    (file "./flow.tcl" line 434)

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 8120faaedf752714e65fb7ff91993a8e6630a664
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

8120faa 2022-05-11T19:42:22+02:00 Make `RUN_SPEF_EXTRACTION` skip STA entirely, remove pandas (#1089) - Mohamed Gaber -  (HEAD -> master, tag: 2022.05.12_01.39.47, origin/master, origin/HEAD)
0c611a3 2022-05-09T08:39:57-07:00 Update to use the new C++ pdngen in OR (#1059) - Matt Liberty -  (tag: 2022.05.11_01.53.47)
47c9f22 2022-05-06T14:32:57+02:00 Remove Hard-coded Paths in Sample Designs (#1083) - Balint Cristian -  (tag: 2022.05.07_01.50.30)
```

### Reproduction Material
* List the commands used to run the design.

cd OpenLane
make mount
./flow.tcl -design ALU

[... 931 lines cut ...]
ABC: + balance 
ABC: + rewrite 
ABC: + rewrite -z 
ABC: + balance 
ABC: + refactor -z 
ABC: + rewrite -z 
ABC: + balance 
ABC: + fraig_store 
ABC: + balance 
ABC: + rewrite 
ABC: + refactor 
ABC: + balance 
ABC: + rewrite 
ABC: + rewrite -z 
ABC: + balance 
ABC: + refactor -z 
ABC: + rewrite -z 
ABC: + balance 
ABC: + fraig_store 
ABC: + fraig_restore 
ABC: + amap -m -Q 0.1 -F 20 -A 20 -C 5000 
ABC: yosys-abc: src/map/amap/amapGraph.c:149: Amap_ManCreateAnd: Assertion `p->nLevelMax < 4094' failed.
ERROR: ABC: execution of command "/build/bin/yosys-abc -s -f /tmp/yosys-abc-NTqtCv/abc.script 2>&1" failed: return code 134.

```

vijayank88, 2022-06-29:
@akifcelik 
Have you tried with latest OpenLane git version?

akifcelik, 2022-06-29:
Hi,
I have run again with different directory but result unchanged and failed in  synthesize.
I have checked following command whether openlane is last version. answer is yes.
          akif@DESKTOP-FA8IJ3J:~/OpenLane$ git checkout master
          Already on 'master'
          Your branch is up to date with 'origin/master'.
          akif@DESKTOP-FA8IJ3J:~/OpenLane$

[image]

Project folder with verilog and config.tcl files:
[Non_storing_divider.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9011888/Non_storing_divider.zip)

vijayank88, 2022-07-01:
@akifcelik 
First thing remove `reg` sv construct from verilog code line 11 & 12 of `divider_64bit_test.v`

Update `config.tcl` with `set ::env(SYNTH_STRATEGY) {DELAY 1}` will pass synthesis stage

akifcelik, 2022-07-02:
@vijayank88 
Hi ,it has also worked with set ::env(SYNTH_STRATEGY) {AREA 3} .thank you for supporting.

closed 2022-07-02 by akifcelik

akifcelik, 2022-10-11:
Hi , ı have latest Openlane version.
I have attached project files to comment on github as a reply.

vijayan ***@***.***>, 29 Haz 2022 Çar, 08:13 tarihinde şunu
yazdı: