The-OpenROAD-Project/OpenLane #1173: PDN hangs with lots of small designs

mattvenn, 2022-06-29:
### Description

PDN hangs forever if a certain combination of designs and PDN spacing is used.

### Environment
```
Kernel: Linux v5.13.0-48-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

### Reproduction Material

* clone https://github.com/mattvenn/tapeout_100
* run 'make user_project_wrapper'

### Expected behavior

PDN shouldn't hang.

### Logs

Last messages:
```
[WARNING PSM-0032] Node at (1571400, 2854575) and layer 5 moved from (1570520, 2655880).
[WARNING PSM-0032] Node at (1512000, 2854575) and layer 5 moved from (1510520, 2655880).
[WARNING PSM-0032] message limit reached, this message will no longer print
[INFO PSM-0064] Number of voltage sources = 88.
```

mattvenn, 2022-06-29:
PDN 90 x 90 seems to work most of the time.

vijayank88, 2022-06-29:
@mattvenn 
As per commit `e4bfdd7` its `mpw-6b` tag.

It had some issues and updated later in efabless to use `mpw-6c` I think.

Also the `caravel_user_project` points `export OPENLANE_TAG=2022.02.23_02.50.41` https://github.com/efabless/caravel_user_project/blob/cd4f35df25e15f6cb35cd3bd777f8779566491e8/Makefile#L26

Its confusing which OpenLane git version used for mpw shuttle.

So updating document will be easy to trace the tool and possible to fix the issue.

I'm suggesting to update OpenLane and try it. With latest tag its not hanging, but getting different error:

```
[STEP 15]
[INFO]: Running Detailed Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/tapeout_100/runs/RUN_2022.06.29_10.54.43/logs/routing/15-detailed.log
[ERROR]: Last 10 lines:
Units:                1000
Number of layers:     13
Number of macros:     442
Number of vias:       25
Number of viarulegen: 25

[INFO DRT-0150] Reading design.
[ERROR DRT-0305] Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets.
Error: droute.tcl, 46 DRT-0305
```

maliberty, 2022-06-29:
@mattvenn please package an issue tarball rather than require running OL as your reproducer.

mattvenn, 2022-06-29:
tar all of caravel_user_project? or something else. The reason I'm not doing that is because with the submodules it's 8GB. Without the submodules it's much less, but then the person running the test has to install them.

maliberty, 2022-06-29:
I mean use [or_issue.py](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md).  I believe it is run automatically when the flow fails.  @donn can help if needed.

vijayank88, 2022-06-30:
@maliberty 
If flow hangs, they will kill it and it will generate `issue_reproducible`.

 Best @mattvenn can use `or_issue.py` to produce testcase  for PDN stage. If any issues let me know, I will help on this.

mattvenn, 2022-06-30:
@vijayank88 it doesn't get killed in a reasonable amount of time. How to generate the test case?

vijayank88, 2022-06-30:
@mattvenn 
problem with pdn stage right.
So add `exit 1` in following line https://github.com/The-OpenROAD-Project/OpenLane/blob/83b61459abcaef3b0ddcf022e1a67e71c88f6007/scripts/openroad/pdn.tcl#L58 and run the flow. It will exit and generate testcase. After this run remove `exit 1` from `pdn.tcl` and continue as it is. Attach the tar file here for test case.

mattvenn, 2022-06-30:
it's hanging before that line, so it still never finishes

mattvenn, 2022-06-30:
so:
* start at the end and move the exit 1 backwards
* wait 10 minutes
* repeat until it exits
?

tbh I'm not likely to complete this process as I have a work around already.

vijayank88, 2022-07-01:
@mattvenn 
Without test case keep this issue open not helpful.

donn, 2022-07-01:
> Use exit 1 before the line where it hangs

No, exit 1 at the start of the script is just fine. Where did you get this brilliant idea?

vijayank88, 2022-07-01:
> > Use exit 1 before the line where it hangs
> 
> No, exit 1 at the start of the script is just fine. Where did you get this brilliant idea?

Copyrights goes to @antonblanchard

donn, 2022-07-01:
Gah. Okay, clearly, what I need to do here is add functionality to just package a specific script without the use of `exit`.

vijayank88, 2022-07-01:
@donn 
With latest OpenROAD commit i am not seeing any hanging problem.
Its discussed here. Maybe @maliberty you can take the PDN stage test from here: https://github.com/The-OpenROAD-Project/OpenLane/issues/1076#issuecomment-1126027261

donn, 2022-07-01:
I'll update OpenROAD and give it a shot then.

maliberty, 2022-07-01:
@vijayank88 if you see no problem then what are you asking me to do?

donn, 2022-07-18:
We've updated to the latest OpenROAD on master. Feel free to re-open if the problem persists.

closed 2022-07-18 by donn