The-OpenROAD-Project/OpenLane #1177: Fail to streaming out gdsii with magic

AyaseErii, 2022-06-29:
### Description
Hello, I am trying to use manual macro placement to generate a small fpga array. However, the flow fails at the stage 31. Please find the error message in the log section. I am not sure what is wrong here. Could anyone point me to the right direction to fix this issue? I will appreciate it! Thanks!

### Environment
```
Kernel: Linux v4.15.0-188-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: a9883335a102c9ba1d862a209a72fc8b4c20c785
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a988333 2022-06-18T11:53:39+02:00 Enhancements + Bugfixes to Multi-PDK Support (#1138) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.19_01.37.45)
732c856 2022-06-18T11:22:57+02:00 Fix `skip_basic_prep`, step mechanics (#1143) - Mohamed Gaber -  ()
074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (tag: 2022.06.15_01.35.48)
```

### Reproduction Material
[RUN_2022.06.29_20.52.26_run_repo.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9014608/RUN_2022.06.29_20.52.26_run_repo.tar.gz)

### Expected behavior
The the entire flow of manual macro placement should be successful.

### Logs
```
[STEP 31]
[INFO]: Running Magic to generate various views...
[INFO]: Streaming out GDS-II with Magic...
[ERROR]: during executing: "magic -noconsole -dnull -rcfile /home/htf6ry/skywater-pdk/sky130A/libs.tech/magic/sky130A.magicrc /openlane/scripts/magic/mag_gds.tcl </dev/null |& tee /dev/null /openlane/designs/fpga_core/runs/RUN_2022.06.29_20.52.26/logs/signoff/31-gdsii.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
   Generating output for cell cbx_1__1_
   Generating output for cell sb_1__1_
   Generating output for cell sky130_fd_sc_hd__buf_12
   Generating output for cell sb_2__1_
   Generating output for cell sb_0__2_
   Generating output for cell cbx_1__2_
   Generating output for cell sb_1__2_
   Generating output for cell sb_2__2_
   Generating output for cell fpga_core
child killed: segmentation violation
```

vijayank88, 2022-06-30:
@AyaseErii 
Child killed probably an out of memory kill. Can you check what is peak memory its used and your system RAM?

vijayank88, 2022-06-30:
@AyaseErii 
issue must be reproducible at out end. The attached test case not contains any required inputs.
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/magic/mag_gds.tcl check this script and attach those required inputs.
Need all lef file and final DEF file to check at our end.

AyaseErii, 2022-06-30:
@vijayank88 Thanks for your help! I happened to check the `.mag` file and found one placement is out of the die area. Because I am using manual macro placement, it looks like there are some miscalculations of the macro spacings causing the strange placement. In other words, the spacings of several macros are too small to place cells and layers. The error has been fixed now after re-calculating the spacing and macro coordinates.

Thanks!

closed 2022-06-30 by AyaseErii