The-OpenROAD-Project/OpenLane #1180: No module named 'odb'

SerdarUnal132, 2022-06-30:
### Description
odb package seems not included in the container. It is called in the padringer.py script but I get "No module named 'odb' error.

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey

python3 ./env.py issue-survey
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 8120faaedf752714e65fb7ff91993a8e6630a664
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

8120faa 2022-05-11T19:42:22+02:00 Make `RUN_SPEF_EXTRACTION` skip STA entirely, remove pandas (#1089) - Mohamed Gaber -  (HEAD -> master, tag: 2022.05.12_01.39.47, origin/master, origin/HEAD)
0c611a3 2022-05-09T08:39:57-07:00 Update to use the new C++ pdngen in OR (#1059) - Matt Liberty -  (tag: 2022.05.11_01.53.47)
47c9f22 2022-05-06T14:32:57+02:00 Remove Hard-coded Paths in Sample Designs (#1083) - Balint Cristian -  (tag: 2022.05.07_01.50.30)
```

### Reproduction Material
* Upload a tarball containing the relevant design.
[chip_io.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9023105/chip_io.tar.gz)

* Place the tar.gz into OpenLane/designs and extract
* (while in OpenLane directory) make mount
* ./flow.tcl -interactive -design chip_io
* source ./designs/chip_io/interactive.tcl

### Expected behavior
odb should have been already set up. padringer.py script is succesfully completed and the script continues

### Logs
```
[STEP 1]
[INFO]: Running Initial Floorplanning...
[INFO]: Setting Core Dimensions...
[INFO]: Generating pad frame
Traceback (most recent call last):
  File "/openlane/scripts/padringer.py", line 29, in <module>
    import odb
ModuleNotFoundError: No module named 'odb'
child process exited abnormally
```

maliberty, 2022-06-30:
@donn I'm not aware of any change in OR to cause this so I guess it is an OL issue.

donn, 2022-07-01:
Hi @SerdarUnal132 -- from what it looks like, you're using a version of OpenLane from May. Can you `git pull` and `make openlane`, please?

SerdarUnal132, 2022-07-01:
I still get the same error after updating the tool

kareefardi, 2022-07-01:
The interactive script is very outdated and won't work with the current openlane without modifications to the script. We never needed to redo chip io so we didn't updated it. Having said I did make some WIP changes to the interactive script here https://github.com/kareefardi/caravel/blob/openlane_config_updates/openlane/chip_io/interactive.tcl . This may not work but I am welcome to try and work it out together.

SerdarUnal132, 2022-07-01:
Thanks for sharing, I will look at it

closed 2022-07-05 by donn