The-OpenROAD-Project/OpenLane #1183: Remove terrible placement configuration variable or output a warning

armleo, 2022-07-03:
### Prompt
In Skywater's Slack you can see people are constantly having issues with the routing. The routing just crashes or errors out. Most likely reasons for this has nothing to do with designs themselved and the errors are very cryptic and hard to debug. For example:

[image]

### Proposal
This is mostly caused by a couple of placement variables: `PL_BASIC_PLACEMENT` `PL_SKIP_INITIAL_PLACEMENT` `PL_RANDOM_GLB_PLACEMENT` `PL_RANDOM_INITIAL_PLACEMENT`. My suggestion would be to at least output a warning at the end of the flow if these variables are set, it would be even better if they are removed altogether. There is no reason to have them in the OpenLane anyway.

maliberty, 2022-07-14:
I upvote this!

mattvenn, 2022-12-05:
They are sometimes needed for small designs though. 
For tiny tapeout we had 160 designs, and 20 of them needed PL_BASIC_PLACEMENT set or they would fail to route.

armleo, 2022-12-05:
Doesnt disabling timing based routing help? Or some other options like increased die-to-core distance?

mattvenn, 2022-12-05:
https://github.com/TinyTapeout/tt02-test-invert
https://github.com/argunda/tt02-TinyPFD
https://github.com/13arn/tt02_counter_steamdeck
https://github.com/NYIT-CNS/cns001-tt02-submission1
https://github.com/shaos/tt02-submission-shaos
https://github.com/mole99/tt02-1bit-alu
https://github.com/jglim/tt02-bcd-7seg
https://github.com/mmolteni-secpat/tinytapeout02_chi3shares
https://github.com/aiunderstand/tt02-4bit-tristate-loadable-counter
https://github.com/ctag/tt02-submission-ctag
https://github.com/AsmaMohsin1507/tt02-channel-coding
https://github.com/ternary-info/tt02-submission-shaos3
https://github.com/saicharan0112/tt02-submission-template
https://github.com/theFestest/tt02-4x4-ram
https://github.com/jeanthom/tinytapout-lock
https://github.com/cmu-stuco-98154/f22-tt02-qilins
https://github.com/prabaldutta/tt02-adi-demo
https://github.com/TinyTapeout/tt02-test-7seg

mattvenn, 2022-12-05:
Is that enough, or you want me to make a new issue that references all these designs?

maliberty, 2022-12-05:
@mattvenn it would be helpful if you could pick one of these and open an issue with a standalone test case.  Once we resolve that we can see how many remain problematic.

mattvenn, 2022-12-08:
opened here: https://github.com/The-OpenROAD-Project/OpenLane/issues/1559

maliberty, 2023-06-23:
Where are the GH issues for "In Skywater's Slack you can see people are constantly having issues with the routing."?