The-OpenROAD-Project/OpenLane #1185: Add insert_tiecells  features to handle special nets.

vijayank88, 2022-07-05:
### Description
If there are unconnected nets, yosys by default connect to low (1'b0). During sta its named as `zero_` or `one_`nets. So getting `Net zero_ of signal type GROUND is not routable by TritonRoute` 

It should be tie to low using `insert_tiecells` feature from OpenROAD. As per comment https://github.com/The-OpenROAD-Project/OpenLane/issues/1002#issuecomment-1142050334 it should be handled during floorplan stage.

Please check how ORFS handle those unconnected nets: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/floorplan.tcl#L107 - 118

Same enhancements should be used in OpenLane flow to handle special nets. 

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: mpw6_4-dev
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

### Reproduction Material
* clone MPW CI:
git clone https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI
cd OpenLane-MPW-CI
./scripts/setup-ci.sh
./scripts/run-design.sh azadi_soc_iii_dft

CI results available here: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Funstable-small-1/detail/main/13/pipeline/387

### Expected behavior
Should complete routing stage.

### Logs
```
[STEP 15]

[INFO]: Running Detailed Routing...

[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl

[ERROR]: Exit code: 1

[ERROR]: full log: designs/azadi_soc_iii_dft/runs/RUN_2022.07.03_15.38.06/logs/routing/15-detailed.log

[ERROR]: Last 10 lines:

Units:                1000

Number of layers:     13

Number of macros:     442

Number of vias:       25

Number of viarulegen: 25

[INFO DRT-0150] Reading design.

[ERROR DRT-0305] Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets.

Error: droute.tcl, 46 DRT-0305

child process exited abnormally

```

vijayank88, 2022-07-15:
@maliberty 
Do we need to add `insert_tiecells` only if there are such special nets or we can use by default after `initialize_floorplan`?

vijayank88, 2022-07-16:
@maliberty 
As per OpenROAD unit test `src/ifp/test/tiecells.tcl` use `insert_tiecells` after  initialize floorplan.

If I use after macro placement TritonRoute fail again with above same error.

maliberty, 2022-07-16:
@vijayank88 sorry I misread it as tapcell - tie cells can be inserted at floorplan time.

donn, 2022-07-22:
@maliberty Does this look right?

```tcl
initialize_floorplan {*}$arg_list

set tielo_cell [lindex $::env(SYNTH_TIELO_PORT) 0]
set tielo_port [lindex $::env(SYNTH_TIELO_PORT) 1]
set tiehi_cell [lindex $::env(SYNTH_TIEHI_PORT) 0]
set tiehi_port [lindex $::env(SYNTH_TIEHI_PORT) 1]

insert_tiecells "$tielo_cell/$tielo_port" -prefix "TIE_ZERO_"
insert_tiecells "$tiehi_cell/$tiehi_port" -prefix "TIE_ONE_"
```

maliberty, 2022-07-22:
@donn looks fine

PR #1226 (2022-07-25): Tie Cells at FP Stage, Improved Cell Padding
\+ Added "logging to" information for long-running steps (CTS and Routing)
\+ Added variable `GPL_CELL_PADDING` for global placement, with a default value of 0 (value divided by 2 and applied to both the left and right of the cell)
\+ Add feature to also print mounted scripts' version if the container version does not match
\+ Add `insert_tiecells` after floorplanning
~ `replace.tcl`, `opendp.tcl` -> `gpl.tcl`, `dpl.tcl`
~ `CELL_PAD` renamed `DPL_CELL_PADDING` for detailed placement, default value still 4, value divided by 2 and applied to both the  [... cut]
closed 2022-07-25 by donn