The-OpenROAD-Project/OpenLane #1188: Skip run_synthesis and use netlist directly with non-interactive mode

vijayank88, 2022-07-07:
### Prompt
Recent designs comes with direct netlist and want to run the flow from floorplan stage.

Seems `interactive` mode directly we can use `set_netlist` to start the `sta` and `floorplan` stage.

### Proposal
Provide variable to pass netlist to floorplan stage and skip `run_synthesis` stage through `non-interactive` mode.

If this feature already exists please give me the steps to follow with `non-interactive` mode.

vijayank88, 2022-07-15:
@donn,
I tested with above setting but its throwing error:
```
[STEP 1]
[INFO]: Running Synthesis...
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/spm/runs/skip_yosys/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
ERROR: TCL interpreter returned an error: can't read "::env(VERILOG_FILES)": no such variable
child process exited abnormally
```
I've used following `config.json`
```
"DESIGN_NAME": "spm",
    "SKIP_SYNTHESIS": 1,
    "CURRENT_NETLIST":"dir::netlist/*.v",
    "CLOCK_PERIOD": 10,
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "ref::$CLOCK_PORT",
    "FP_PDN_AUTO_ADJUST": 0,
    "FP_PDN_VOFFSET": 7,
    "FP_PDN_HOFFSET": 7,
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg",
```

vijayank88, 2022-07-15:
Also did `grep -ir "SKIP_SYNTHESIS" scripts/*` no such configuration found.

vijayank88, 2022-07-15:
@donn 
Seems following tweak in the scripts fix this: https://github.com/The-OpenROAD-Project/OpenLane/blob/c38101ea8add2c08746741bb6c71a7d320660e37/scripts/tcl_commands/synthesis.tcl#L112
```
proc run_synthesis {args} {
    increment_index
    TIMER::timer_start
    puts_info "Running Synthesis..."
    set ::env(CURRENT_SDC) $::env(BASE_SDC_FILE)
    # in-place insertion
    #if { [file exists $::env(synthesis_results)/$::env(DESIGN_NAME).v] } {}#
    if { $::env(SKIP_SYNTHESIS) } {
        file copy $::env(DESIGN_DIR)/netlist/$::env(DESIGN_NAME).v $::env(synthesis_results)
        puts_warn "A netlist at $::env(synthesis_results)/$::env(DESIGN_NAME).v already exists. Synthesis will be skipped."
        set_netlist $::env(synthesis_results)/$::env(DESIGN_NAME).v
    } else {
        run_yosys
    }
```

donn, 2022-07-15:
That's overkill.

Try `./flow.tcl -from floorplan -design whatever`?

vijayank88, 2022-07-15:
By default its looking `runs/RUN_2022.07.15_13.13.40/results/synthesis/spm.v` to start floorplan.
How do we place that file to that specific path?
After execution only `RUN_2022.07.15_13.13.40` this directory will be created

kareefardi, 2022-07-15:
how about
```
set ::env(SYNTH_READ_BLACKBOX_LIB) 1
set ::env(SYNTH_TOP_LEVEL) 1
```
but you should also provide `VERILOG_FILES` which I guess in your case is a single netlist. it is not exactly as skipping synthesis as the netlist will go through yosys but if the nelist is correct the logic shouldn't be touched.

vijayank88, 2022-07-15:
In that case I should pass netlist as VERILOG_FILES?

Because I don't have source RTL.

maliberty, 2022-07-15:
You can use -tag to control the rundir name

vijayank88, 2022-07-16:
> SYNTH_READ_BLACKBOX_LIB

@kareefardi Thanks. This option is working.
Found document from here: https://github.com/The-OpenROAD-Project/OpenLane/blob/b29b71536b644dd76550dfdfcc09319b4ed86a79/docs/source/chip_integration.md#hardening-the-full-chip
Thanks.

closed 2022-07-16 by vijayank88

kareefardi, 2022-07-18:
@vijayank88 Great. Just curious, did you also enable `SYNTH_TOP_LEVEL`?

vijayank88, 2022-07-18:
yes. 
```
set ::env(SYNTH_READ_BLACKBOX_LIB) 1
set ::env(SYNTH_TOP_LEVEL) 1
```