The-OpenROAD-Project/OpenLane #1191: UnicodeDecodeError when executing a python script

GabbedT, 2022-07-10:
Hi, I was trying to synthesize a design when at the end of the flow basically crashed. It was at the 39-th step, this is the output:
```
[STEP 39]
[INFO]: Running CVC...
[INFO]: Saving current set of views in 'designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/results/final'...
[INFO]: Saving runtime environment...
[INFO]: Generating final set of reports...
[ERROR]: during executing: "openroad -python /openlane/scripts/generate_reports.py -d /openlane/designs/pipelined_long_multiplier --design_name pipelined_long_multiplier --tag RUN_2022.07.10_10.22.17 --output_file /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/metrics.csv --man_report /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/manufacturability.rpt --run_path /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openlane/scripts/report/report.py", line 758, in get_report
    *self.extract_all_values(),
  File "/openlane/scripts/report/report.py", line 698, in extract_all_values
    cvc_log_content = cvc_log.get_content()
  File "/openlane/scripts/report/report.py", line 94, in get_content
    return open(self.path).read()
  File "/usr/lib64/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 2517: invalid start byte <----- ERROR

[INFO]: Saving current set of views in 'designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/results/final'...
[INFO]: Generating final set of reports...
[ERROR]: during executing: "openroad -python /openlane/scripts/generate_reports.py -d /openlane/designs/pipelined_long_multiplier --design_name pipelined_long_multiplier --tag RUN_2022.07.10_10.22.17 --output_file /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/metrics.csv --man_report /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/manufacturability.rpt --run_path /openlane/designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openlane/scripts/report/report.py", line 758, in get_report
    *self.extract_all_values(),
  File "/openlane/scripts/report/report.py", line 698, in extract_all_values
    cvc_log_content = cvc_log.get_content()
  File "/openlane/scripts/report/report.py", line 94, in get_content
    return open(self.path).read()
  File "/usr/lib64/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 2517: invalid start byte <----- ERROR

[INFO]: Created manufacturability report at 'designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/pipelined_long_multiplier/runs/RUN_2022.07.10_10.22.17/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -python $::env(OPENLANE_ROOT)/scripts/generate_reports.py -d $::env(DESIGN_DIR)  --design_name $::env(DESIGN_NAME)  --t..."
    (procedure "generate_final_summary_report" line 16)
    invoked from within
"generate_final_summary_report"
    (procedure "run_non_interactive_mode" line 81)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 384)
```
 
I also have the same problem when running a configuration exploration:
```
python3 run_designs.py pipelined_long_multiplier
[2022-07-10 10:11:12,869 -  INFO] [START] pipelined_long_multiplier
[2022-07-10 10:16:53,907 - ERROR] [FAIL ] pipelined_long_multiplier: Check designs/pipelined_long_multiplier/runs/config_regression/openlane.log
[2022-07-10 10:16:53,907 -  INFO] [DONE ] pipelined_long_multiplier: Writing report...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "run_designs.py", line 268, in run_design
    report = Report(design, tag, design_name, params).get_report()
  File "/openlane/scripts/report/report.py", line 758, in get_report
    *self.extract_all_values(),
  File "/openlane/scripts/report/report.py", line 698, in extract_all_values
    cvc_log_content = cvc_log.get_content()
  File "/openlane/scripts/report/report.py", line 94, in get_content
    return open(self.path).read()
  File "/usr/lib64/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 2475: invalid start byte <------ ERROR

[2022-07-10 10:16:55,332 -  INFO] Exiting thread 0...
[2022-07-10 10:16:55,332 -  INFO] Getting top results...
[2022-07-10 10:16:55,358 -  INFO] Done.
```
I have updated `python3` to 3.10.5 but when i run `python3 --version` inside docker container it gives me 3.6.8. I don't know if that is the problem.

This is the environment:
```
gabriele@Gabbed:~/OpenLane$ make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v5.13.0-52-generic
Distribution: ubuntu 20.04
Python: v3.10.5 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 074a92b7895be96a66db4318015ecb9f2c0f93c2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.15_01.35.48)
5a767e7 2022-06-10T15:28:44+02:00 Updates to `save_final_views` (#1137) - Mohamed Gaber -  (tag: 2022.06.11_01.51.35)
ad24025 2022-06-07T08:52:04+02:00 Replace -slack_margin with -setup_margin/-hold_margin (#1130) - Anton Blanchard -  (tag: 2022.06.08_01.54.04)
```

Thanks for helping me.

vijayank88, 2022-07-11:
@GabbedT 
Have you find any `issue_reproducible` inside run directory?

If not please package a test case and attach. share your updated repo link.

GabbedT, 2022-07-11:
@vijayank88 No, i can't find any, however I tried to execute the flow with other design and it completed without any problem.
Do I need to upload only the folder of the design that gives me the error, or the entire OpenLane repository?

vijayank88, 2022-07-11:
@GabbedT 

Is your design contains multiple power domains?

vijayank88, 2022-07-11:
@GabbedT 
While generating final reports its throws an error. Need to check the reason for the crash. Can you share your repo link?

GabbedT, 2022-07-11:
@vijayank88 

> @GabbedT
> 
> Is your design contains multiple power domains?

No, it has only one power domain. 

> @GabbedT While generating final reports its throws an error. Need to check the reason for the crash. Can you share your repo link?

I forked the original OpenLane repository, what should i do? How can i push the entire OpenLane folder on the disk into the fork?

vijayank88, 2022-07-12:
@GabbedT 
`designs/pipelined_long_multiplier/src` and `config.tcl` just zip those and share it

GabbedT, 2022-07-12:
@vijayank88 https://github.com/GabbedT/OpenLane/tree/master/designs/pipelined_long_multiplier

donn, 2022-07-14:
So, preliminary investigation shows that, for some reason, CVC just outputs garbage bytes in its logs while working with this design:

[image]

As a rule, cvc is an inessential check, so you can bypass this issue by setting `RUN_CVC` to 0 while we investigate further with @d-m-bailey.

d-m-bailey, 2022-07-14:
Looks like `sky130_ef_sc_hd__decap_12` is not included in `sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl`.

Possible solutions:
1. Add a filter that converts `sky130_ef` to `sky130_fd` when creating the CDL netlist for CVC.
2. Add `sky130_ef_sc_hd__decap_12` to `sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl`.
3. Create a temporary library with `sky130_ef_sc_hd__decap_12` merged with `sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl`.
4. Remove the `sky130_??_sc_*__` prefixes from the cdl netlists.

vijayank88, 2022-07-16:
@GabbedT 
Have you tried with `set ::env(RUN_CVC) 0` configuration?

GabbedT, 2022-07-16:
@vijayank88 @donn Yes, now it works. I also tried the 4-th solution of @d-m-bailey but it didn't work here. I don't understand how to proceed with the other solutions.

donn, 2022-07-17:
Yeah, that's a workaround. The rest we'd have to implement in some capacity in OpenLane.

@d-m-bailey That's understandable but I still don't quite get why cvc would be printing garbage data at the end.

mattvenn, 2022-11-29:
Just hit this with https://github.com/ARamsey118/tiny_tapeout_freq_counter

d-m-bailey, 2022-11-29:
@mattvenn Can you add the CVC log files to `.gitignore` so they don't get pushed to the repo?

mattvenn, 2022-11-29:
yeah, that's why I've done

kareefardi, 2023-09-10:
CVC is no longer support in OpenLane flow.

closed 2023-09-10 by kareefardi

d-m-bailey, 2023-09-11:
Just to be clear. CVC has been moved from openlane flow to mpw_precheck.