The-OpenROAD-Project/OpenLane #1192: Corrupt SDF file (duplicate MUX IOPATH S X paths with different values)

cloudxcc, 2022-07-10:
### Description
The S-> X IOPATH of MUXes are duplicated in the SDF file and have different values:

SDF file snipset:

 (CELL
  (CELLTYPE "sky130_fd_sc_hd__mux2_2")
  (INSTANCE _344_)
  (DELAY
   (ABSOLUTE
    (IOPATH A0 X (0.159:0.159:0.159) (0.268:0.268:0.268))
    (IOPATH A1 X (0.125:0.125:0.125) (0.243:0.243:0.243))
    (IOPATH S X (0.233:0.233:0.233) (0.341:0.341:0.341))
    (IOPATH S X (0.163:0.163:0.163) (0.279:0.279:0.279))  <--- second arc with > 60 ps difference
   )
  )
 )

This is most likely true for all MUXes, but not for other cells.

### Environment
Tested on latest mpw-7a (but seen it  before):
OpenLane f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
PDK version 41c0908b47130d5675ff8484255b43f66463a7d6

### Reproduction Material
Latest caravel example:
git clone -b mpw-7a https://github.com/efabless/caravel_user_project caravel_example

### Expected behavior
Only one entry per arc.

PS: also wondering why bc:nc:wc numbers are identical at many cells.

jjcherry56, 2022-07-10:
Those are IOPATHs with different COND expressions that for some reason are not printed in the SDF.
The bc:typ:wc values are the same because there is only one process corner and the no min/max slew spread.
Without an explicit testcase (ie, a tcl command file that produces the SDF) I can't help you any further.

cloudxcc, 2022-07-10:
The explicit testcase is given in the text above. Use the actual, mpw-7a, default caravel_example project.

git clone -b mpw-7a https://github.com/efabless/caravel_user_project caravel_example

The earliest SDF file I have quick access to is from Dec '21 where I was fooling around with MPW-3. Same problem there, so I guess you can find that issue on any given Openlane SDF file.

Openlane SDF file supports different COND conditions ?

maliberty, 2022-07-10:
The expectation is that you will package a standalone test case.  See https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

@donn can help if required.

jjcherry56, 2022-07-10:
It looks like the issue is the sky130 liberty.
cell sky130_fd_sc_hd__mux2_1 has these timing arcs from S -> X
        pin ("X") {
            direction : "output";
            function : "(A0&!S) | (A1&S)";
            timing () {
                related_pin : "S";
                timing_sense : "positive_unate";
                timing_type : "combinational";
            }
            timing () {
                related_pin : "S";
                timing_sense : "negative_unate";
                timing_type : "combinational";
            }
        }

They should have sdf_cond attributes to label them for the SDF.

cloudxcc, 2022-07-11:
@jjcherry56 Conditions in the SDF file ? Is this possible ?

So I guess that both path are valid and my STA engine should take the most critical one, right ?

maliberty, 2022-07-11:
There is syntax to support it in [SDF](https://web.archive.org/web/20131104211358/http://www.eda.org/sdf/sdf_3.0.pdf):
[image]

The timing engine will select the right arc.  However your SDF is not valid because it lacks the COND statements for the reason Cherry describes above.

maliberty, 2022-07-11:
I opened https://github.com/google/skywater-pdk/issues/382 for this issue.

@mithro @quantumhd please advise if you will fix this in the upstream repo or not

mithro, 2022-07-12:
Unlikely to happen in the next few weeks. Best bet is to add something to open_pdks temporarily.

donn, 2022-07-13:
@RTimothyEdwards You're up.