The-OpenROAD-Project/OpenLane #1193: no such variable error while preparing LEF files

anilberg, 2022-07-10:
### Description
Hi all, I am trying to learn design with OpenLane flow. After I build the full pdk packet, this error occured during the OpenLane flow.

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-39-generic
Distribution: ubuntu 22.04
Python: v3.10.4 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: b718fd064063333703734ce3ae8bb56aa6810b0b
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b718fd0 2022-07-05T10:05:16+03:00 Remove stale comment about power pin and net being the same (#1184) - Anton Blanchard -  (HEAD -> master, tag: 2022.07.06_01.50.05, origin/master, origin/HEAD)
b63f46b 2022-07-05T08:35:35+03:00 Added missing `-log` to `scrot_klayout` call (#1182) - Syndace -  ()
f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (tag: 2022.07.02_01.38.08)
```

### Reproduction Material
My design and config file.
[verilog&config_file.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9080087/verilog.config_file.tar.gz)

### Expected behavior
There was no error in the same flow before I build full pdk packet.

### Logs
OpenLane Flow output
```
[INFO]: Using design configuration at /openlane/designs/mydesigns/VLSI-Workshop/Adders/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/mydesigns/VLSI-Workshop/Adders/config.tcl
[INFO]: PDKs root directory: /home/anilberg/Desktop/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/anilberg/Desktop/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/mydesigns/VLSI-Workshop/Adders/config.tcl
[INFO]: Run Directory: /openlane/designs/mydesigns/VLSI-Workshop/Adders/runs/RUN_2022.07.10_22.50.40
[INFO]: Preparing LEF files for the nom corner...
no such variable
    (read trace on "::env(DRC_EXCLUDE_CELL_LIST)")
    invoked from within
"set_if_unset arg_values(-drc_exclude_list) $::env(DRC_EXCLUDE_CELL_LIST)"
    (procedure "gen_exclude_list" line 15)
    invoked from within
"gen_exclude_list -lib $arg_values(-output)"
    (procedure "trim_lib" line 17)
    invoked from within
"trim_lib -input $::env(LIB_SYNTH_MERGED)"
    (procedure "prep" line 285)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 389)
```

vijayank88, 2022-07-11:
@anilberg 
have you tried other libraries?

anilberg, 2022-07-11:
I've already tried with default library `set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hd"`
Even If I run `make test` , the same error still occurs.

`make test` output:

```
[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/anilberg/Desktop/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/anilberg/Desktop/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: Run Directory: /openlane/designs/spm/runs/openlane_test
[INFO]: Removing existing /openlane/designs/spm/runs/openlane_test...
[INFO]: Preparing LEF files for the nom corner...
no such variable
    (read trace on "::env(DRC_EXCLUDE_CELL_LIST)")
    invoked from within
"set_if_unset arg_values(-drc_exclude_list) $::env(DRC_EXCLUDE_CELL_LIST)"
    (procedure "gen_exclude_list" line 15)
    invoked from within
"gen_exclude_list -lib $arg_values(-output)"
    (procedure "trim_lib" line 17)
    invoked from within
"trim_lib -input $::env(LIB_SYNTH_MERGED)"
    (procedure "prep" line 285)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 389)
make: *** [Makefile:158: test] Error 1
```

vijayank88, 2022-07-11:
@anilberg 
Can you copy and paste the `config.tcl` here?

anilberg, 2022-07-11:
config.tcl of spm:

```
# Design
set ::env(DESIGN_NAME) "spm"

set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

set ::env(CLOCK_PERIOD) "10.000"
set ::env(CLOCK_PORT) "clk"

set ::env(FP_PDN_AUTO_ADJUST) 0
set ::env(FP_PDN_VOFFSET) 7.0
set ::env(FP_PDN_HOFFSET) 7.0

set ::env(FP_PIN_ORDER_CFG) $::env(DESIGN_DIR)/pin_order.cfg

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
```

vijayank88, 2022-07-11:
Do `make mount` followed by `./flow.tcl -design spm` try this and let me know

anilberg, 2022-07-11:
Still the same output

```
anilberg@asus:~/Desktop/OpenLane$ sudo make mount
[sudo] password for anilberg: 
cd /home/anilberg/Desktop/OpenLane && \
	docker run --rm -v /home/anilberg/Desktop/OpenLane:/openlane -v /home/anilberg/Desktop/OpenLane/designs:/openlane/install -v /home/anilberg/Desktop/OpenLane/pdks:/home/anilberg/Desktop/OpenLane/pdks -e PDK_ROOT=/home/anilberg/Desktop/OpenLane/pdks -e PDK=sky130A -e STD_CELL_LIBRARY=sky130_fd_sc_hd  --user 0:0 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /root/.Xauthority:/.Xauthority --network host -ti efabless/openlane:b718fd064063333703734ce3ae8bb56aa6810b0b-amd64
OpenLane Container (b718fd0):/openlane$ ./flow.tcl -design spm
OpenLane b718fd064063333703734ce3ae8bb56aa6810b0b
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/anilberg/Desktop/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/anilberg/Desktop/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: Run Directory: /openlane/designs/spm/runs/RUN_2022.07.11_11.53.53
[INFO]: Preparing LEF files for the nom corner...
no such variable
    (read trace on "::env(DRC_EXCLUDE_CELL_LIST)")
    invoked from within
"set_if_unset arg_values(-drc_exclude_list) $::env(DRC_EXCLUDE_CELL_LIST)"
    (procedure "gen_exclude_list" line 15)
    invoked from within
"gen_exclude_list -lib $arg_values(-output)"
    (procedure "trim_lib" line 17)
    invoked from within
"trim_lib -input $::env(LIB_SYNTH_MERGED)"
    (procedure "prep" line 285)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 389)
```

vijayank88, 2022-07-11:
If you've enough space install OpenLane again in new directory and try again. Because its a setup issue at your end.

anilberg, 2022-07-11:
@vijayank88 Yes, I reinstall OpenLane and It worked when I ran `make test`. But I tried to build other pdks by following the instructions in the page below:
[https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/manual_pdk_installation.md](url)

I ran this command
```
sudo ./configure --enable-sky130-pdk=/home/anilberg/Desktop/OpenLane/pdks/skywater-pdk/libraries --with-sky130-local-path=/home/anilberg/Desktop/OpenLane/pdks --enable-sram-sky130=disabled
```

And output was shown below:
```
...
  from distutils.version import LooseVersion
Traceback (most recent call last):
  File "/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/../common/foundry_install.py", line 430, in <module>
    sourcedir = option[1]
IndexError: list index out of range
# Install OSU T18 ms digital standard cells.
set -f ; ../common/foundry_install.py -std_format -source  -target /home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/sky130A \
	-techlef sky130_osu_sc_18T.tlef rename=sky130_osc_sc_18t_ms.tlef \
	-spice 18T_ms/cdl/*.spice compile-only exclude=*.*.spice \
	-lef 18T_ms/lef/*.lef \
	-lib 18T_ms/lib/*.lib \
	-gds 18T_ms/gds/*.gds compile-only \
		options=custom/scripts/gds_import_setup.tcl \
	-library digital sky130_osu_sc_18t_ms 2>&1 | tee -a sky130A_make.log
/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/../common/foundry_install.py:193: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import LooseVersion
Traceback (most recent call last):
  File "/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/../common/foundry_install.py", line 430, in <module>
    sourcedir = option[1]
IndexError: list index out of range
# Install OSU T18 ls digital standard cells.
set -f ; ../common/foundry_install.py -std_format -source  -target /home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/sky130A \
	-techlef sky130_osu_sc_18T.tlef rename=sky130_osc_sc_18t_ls.tlef \
	-spice 18T_ls/cdl/*.spice compile-only exclude=*.*.spice \
	-lef 18T_ls/lef/*.lef \
	-lib 18T_ls/lib/*.lib \
	-gds 18T_ls/gds/*.gds compile-only \
		options=custom/scripts/gds_import_setup.tcl \
	-library digital sky130_osu_sc_18t_ls 2>&1 | tee -a sky130A_make.log
/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/../common/foundry_install.py:193: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import LooseVersion
Traceback (most recent call last):
  File "/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130/../common/foundry_install.py", line 430, in <module>
    sourcedir = option[1]
IndexError: list index out of range
make[1]: Leaving directory '/home/anilberg/Desktop/OpenLane/pdks/open_pdks/sky130'
echo "Ended sky130A PDK staging on "`date` >> sky130A_make.log
```

I think this build stage causes the crash the OpenLane.

vijayank88, 2022-07-12:
Required PDKs already installed via volare for openlane flow. What's other
PDK you are trying to install?

vijayank88, 2022-07-16:
@donn 
`sky130_fd_sc_hs` is missing from libraries list. Is that removed permanently?

donn, 2022-07-18:
No, it's just not included with volare.

donn, 2022-07-18:
@anilberg, I added a new method of building the PDK that is more... stable. It's been pushed to master. Could you give it a shot?

closed 2023-01-18 by anilberg