The-OpenROAD-Project/OpenLane #1194: Simulation fails and i get this error message ," routing congestion too high"

binoy01, 2022-07-12:
Hi,
    flow failed due to " routing congestion too high". What could be the reason for this? Please check the attachments.
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9090051/issue_reproducible.zip)
[image]

vijayank88, 2022-07-12:
@binoy01 
as per slack channel conversation:  https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1657536454239019
You're using default `config.tcl` from `user_proj_example` with die area `0 0 900 600` is not sufficient to perform optimization. See the congestion below:
[image]

So update `config.tcl` with following variables and run the flow again.
```
# SPDX-FileCopyrightText: 2020 Efabless Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#      http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
# SPDX-License-Identifier: Apache-2.0

set ::env(PDK) $::env(PDK)
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hd"

set script_dir [file dirname [file normalize [info script]]]

set ::env(DESIGN_NAME) prediction

set ::env(VERILOG_FILES) "\
	$::env(CARAVEL_ROOT)/verilog/rtl/defines.v \
	$script_dir/../../verilog/rtl/prediction.v"

set ::env(DESIGN_IS_CORE) 0

set ::env(CLOCK_PORT) "wb_clk_i"
#set ::env(CLOCK_NET) "counter.clk"
set ::env(CLOCK_PERIOD) "10"

#set ::env(FP_SIZING) absolute
#set ::env(DIE_AREA) "0 0 900 600"

set ::env(FP_PIN_ORDER_CFG) $script_dir/pin_order.cfg

set ::env(FP_CORE_UTIL) 30
set ::env(PL_BASIC_PLACEMENT) 0
set ::env(PL_TARGET_DENSITY) 0.40
set ::env(CELL_PAD) 2

# Maximum layer used for routing is metal 4.
# This is because this macro will be inserted in a top level (user_project_wrapper) 
# where the PDN is planned on metal 5. So, to avoid having shorts between routes
# in this macro and the top level metal 5 stripes, we have to restrict routes to metal4.  
# 
set ::env(RT_MAX_LAYER) {met4}

# You can draw more power domains if you need to 
set ::env(VDD_NETS) [list {vccd1}]
set ::env(GND_NETS) [list {vssd1}]

set ::env(DIODE_INSERTION_STRATEGY) 4 
# If you're going to use multiple power domains, then disable cvc run.
set ::env(RUN_CVC) 1
```

binoy01, 2022-07-12:
> @binoy01 Missing issue template. Run the following and update here: `make survey || python3 ./env.py issue-survey`

From which location this has to be run?

binoy01, 2022-07-12:
> @binoy01 as per slack channel conversation: https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1657536454239019 You're using default `config.tcl` from `user_proj_example` with die area `0 0 900 [...]

This one also failed.

vijayank88, 2022-07-12:
can you attache the `issue_reproducible` generated with latest run?

vijayank88, 2022-07-12:
> @binoy01 Missing issue template. Run the following and update here: `make survey || python3 ./env.py issue-survey`

Run  it where you installed OpenLane

binoy01, 2022-07-12:
@vijayank88 
[image]

vijayank88, 2022-07-12:
@binoy01 
Sorry for the confusion.
Just modify following value and run again
```
set ::env(PL_TARGET_DENSITY) 0.55
```

vijayank88, 2022-07-12:
@binoy01 
Can you attach `openlane/prediction/runs/22_07_12_15_12/config.tcl`?

binoy01, 2022-07-12:
@vijayank88 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9091862/issue_reproducible.zip)

This is the status after changing target density to 0.55.     
[image]

vijayank88, 2022-07-12:
@binoy01 
Seems still you're using old `config.tcl` and not yet updated to shared by me

binoy01, 2022-07-12:
@vijayank88 
I have updated the config.tcl file.

[image]

vijayank88, 2022-07-12:
@binoy01 Can you confirm is same as following one?
```
# SPDX-FileCopyrightText: 2020 Efabless Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#      http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
# SPDX-License-Identifier: Apache-2.0

set ::env(PDK) $::env(PDK)
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hd"

set script_dir [file dirname [file normalize [info script]]]

set ::env(DESIGN_NAME) prediction

set ::env(VERILOG_FILES) "\
	$::env(CARAVEL_ROOT)/verilog/rtl/defines.v \
	$script_dir/../../verilog/rtl/prediction.v"

set ::env(DESIGN_IS_CORE) 0

set ::env(CLOCK_PORT) "wb_clk_i"
#set ::env(CLOCK_NET) "counter.clk"
set ::env(CLOCK_PERIOD) "10"

#set ::env(FP_SIZING) absolute
#set ::env(DIE_AREA) "0 0 900 600"

set ::env(FP_PIN_ORDER_CFG) $script_dir/pin_order.cfg

set ::env(FP_CORE_UTIL) 30
set ::env(PL_BASIC_PLACEMENT) 0
set ::env(PL_TARGET_DENSITY) 0.40
set ::env(CELL_PAD) 2

# Maximum layer used for routing is metal 4.
# This is because this macro will be inserted in a top level (user_project_wrapper) 
# where the PDN is planned on metal 5. So, to avoid having shorts between routes
# in this macro and the top level metal 5 stripes, we have to restrict routes to metal4.  
# 
set ::env(RT_MAX_LAYER) {met4}

# You can draw more power domains if you need to 
set ::env(VDD_NETS) [list {vccd1}]
set ::env(GND_NETS) [list {vssd1}]

set ::env(DIODE_INSERTION_STRATEGY) 4 
# If you're going to use multiple power domains, then disable cvc run.
set ::env(RUN_CVC) 1
```

vijayank88, 2022-07-12:
```
#set ::env(FP_SIZING) absolute
#set ::env(DIE_AREA) "0 0 900 600"
```
above lines begin with `#` i.e commented, but in your file its not commented. Its keep on throwing same error.

closed 2022-07-13 by vijayank88