The-OpenROAD-Project/OpenLane #1195: conda-eda issue about antenna

hello-eternity, 2022-07-12:
I meet some error when using conda-eda which uses the python script in this repo, could I issue here?
The error is cause by scripts/extract_antenna_count.py, when Fastrouitng stage. The test case is spm and the config.tcl is 
```
set ::env(DESIGN_NAME) spm

set script_dir [file dirname [file normalize [info script]]]
set ::env(VERILOG_FILES) "$script_dir/spm.v"
 
set ::env(CLOCK_TREE_SYNTH) 0
set ::env(PL_TARGET_DENSITY) 1
set ::env(CLOCK_PORT) "clk"

set ::env(FP_CORE_UTIL) 20
set ::env(RUN_CVC) 0
set ::env(RUN_KLAYOUT) 0

```
The syntax where the error occurs is this sentence. If you delete this sentence, no error will be reported.
`print(re.findall(r"[Aa]ntenna violations:\s*(\d+)", in_data)[-1], end="")`

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

python3 ./env.py issue-survey
Kernel: Linux v5.4.0-115-generic
Distribution: ubuntu 20.04
Python: v3.7.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -> master, tag: 2022.05.18_02.12.32)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()
If neither command succeeds, you are using an out of date version of OpenLane and should probably update. 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.

[STEP 16]
[INFO]: Running Global Routing...
[INFO]: Starting FastRoute Antenna Repair Iterations...
Traceback (most recent call last):
  File "/eda/condaeda/OpenLane/scripts/extract_antenna_count.py", line 6, in <module>
    print(re.findall(r"[Aa]ntenna violations:\s*(\d+)", in_data)[-1], end="")
IndexError: list index out of range
    while executing
"exec python3 $::env(SCRIPTS_DIR)/extract_antenna_count.py < $arg_values(-from_log)"
    (procedure "groute_antenna_extract" line 8)
    invoked from within
"groute_antenna_extract -from_log [index_file $::env(routing_logs)/global.log]"
    (procedure "global_routing_fastroute" line 19)
    invoked from within
"global_routing_fastroute"
    (procedure "global_routing" line 14)
    invoked from within
"global_routing"
    (procedure "run_routing" line 34)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 57)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "/eda/condaeda/OpenLane/flow.tcl" line 389)
```
The run.sh script is 
```
export CONDA_ROOT=/root/anaconda3/envs/vizier/
export PDK_ROOT=$CONDA_ROOT/share/pdk
export TCLLIBPATH=$CONDA_ROOT/lib/tcllib1.20
export OPENLANE_ROOT=/eda/condaeda/OpenLane
export PATH=$PATH:$OPENLANE_ROOT:$OPENLANE_ROOT/scripts
export OPENLANE_LOCAL_INSTALL=1
export TEST_MISMATCHES=none
flow.tcl -design .
```

vijayank88, 2022-07-12:
@hello-eternity 
I think this issue got fixed with latest release.
[2022.07.06_01.50.05](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.07.06_01.50.05)
Can you please update your repo and try once.

proppy, 2022-07-12:
@vijayank88 can you point to the commits that fixed this issue? (I still get it on OpenLane master)

proppy, 2022-07-12:
Since this script is triggered by `DIODE_INSERTION_STRATEGY==3`:
https://github.com/The-OpenROAD-Project/OpenLane/blob/ebad315d1def25d9d253eb2ec1c56d7b4e59d7ca/scripts/tcl_commands/routing.tcl#L67
setting it to another value could be a temporary workaround to skip that step.

proppy, 2022-07-12:
however setting `DIODE_INSERTION_STRATEGY=0` ends up triggering the following error later in the flow:
```
[INFO]: Running OpenROAD Antenna Rule Checker...
[ERROR]: during executing: "openroad -python /OpenLane/scripts/extract_antenna_violators.py --output /OpenLane/designs/inverter/runs/RUN_2022.07.12_15.44.38/reports/signoff/34-antenna_violators.rpt /OpenLane/designs/inverter/runs/RUN_2022.07.12_15.44.38/reports/signoff/34-antenna.rpt"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
    return self.main(*args, **kwargs)
  File "/opt/conda/envs/silicon/lib/python3.7/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/opt/conda/envs/silicon/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/conda/envs/silicon/lib/python3.7/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/OpenLane/scripts/extract_antenna_violators.py", line 34, in extract_antenna_violators
    with open(report, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/OpenLane/designs/inverter/runs/RUN_2022.07.12_15.44.38/reports/signoff/34-antenna.rpt'
```

hello-eternity, 2022-07-13:
Thank you, I temporary do this, and this worked. @proppy 
I update to the latest version and find it did not sovle this. @vijayank88

vijayank88, 2022-07-16:
> @hello-eternity This issue got fixed with latest release. [2022.07.06_01.50.05](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.07.06_01.50.05) Can you please update your repo and try once?

@proppy Have you tried with the latest tag?
@donn  FYI

donn, 2022-11-30:
Staled out.

closed 2022-11-30 by donn