The-OpenROAD-Project/OpenLane #1197: remove PL_RANDOM_GLB_PLACEMENT

maliberty, 2022-07-12:
### Prompt
PL_RANDOM_GLB_PLACEMENT is a mis-feature.  The docs say
```
 This is useful if the design is tiny (less than 100 cells). 
```
I don't think it is needed for any size design but recently @proppy was using it on a very large design and getting poor results.  If a small design can't be placed by OR then file a bug.  This option should be removed as it is dangerous and unhelpful.

### Proposal
Remove PL_RANDOM_GLB_PLACEMENT

donn, 2022-07-14:
Duplicate of #1183- please comment your proposal there.

In principle, I'm not opposed to this if placing small designs with OpenROAD works now, but I recall simple tests like the inverter going absolutely haywire last I checked.

closed 2022-07-14 by donn

maliberty, 2022-07-14:
@donn if that is the case file an issue