The-OpenROAD-Project/OpenLane #1200: make test complains about no module name click even though it is in venv + podman support

jkeltz, 2022-07-13:
Hi.
I have two issues..

1) When running "make test", I see:

```
Traceback (most recent call last):
  File "/tmp/OpenLane/./docker/utils.py", line 18, in <module>
    import click
ModuleNotFoundError: No module named 'click'
```

Yes, click module is required, and in fact, it is even installed in the venv, but rather than calling:

```
% venv/bin/python3 ./docker/utils.py current-docker-platform
amd64
```
... it is instead calling:

```
 % python3 ./docker/utils.py current-docker-platform
Traceback (most recent call last):
  File "/tmp/OpenLane/./docker/utils.py", line 18, in <module>
    import click
ModuleNotFoundError: No module named 'click'
```

I tried installing click in the main python install, and it's fine with that, but I suspect this is a bug. 

Second, and probably less important -- I'm installing on RHEL8 where podman is installed instead of Docker.  podman is command line equivalent to docker.  Everything works fine if I symlink "docker" to "podman".  However, it would be useful for the installer to just be able to detect podman, and just call on that instead of docker.  It would make running OpenLane on this platform just one step easier.

vijayank88, 2022-07-14:
@jkeltz 
Missing issue template:
`make survey || python3 ./env.py issue-survey`

donn, 2022-07-14:
@vijayank88 Not necessary.

---

@jkeltz Thanks for noticing the python3 issue- I'm on it.

As for the podman -> docker thing, sorry, I don't think I want to do that. On paper we only support Docker, and while I have used Podman personally in the past, I do not want to endorse it as a "verified" method of running OpenLane. There are some subtle differences to how Docker and Containerd-based solutions work (and can be configured.) IMO, having a user symlink `podman` to `docker` is a nice, organic "I know what I'm doing" statement, but of course, I welcome any further thoughts on the matter.

vijayank88, 2022-07-14:
@donn 
is this related to RHEL8 OS? 
Even i seen someone reported with Ubuntu 22.04.

donn, 2022-07-14:
No, none of this is necessarily exclusive to RHEL and its derivatives. Podman is the preferred system for running containers on RHEL over Docker, but I've used it for about half a year on Arch Linux without issue.

jkeltz, 2022-07-14:
@donn 
Thanks for looking into the Makefile issue.
Re: podman - obviously, I would love if the support was added because it makes things easier for me, but I respect that you don't want to endorse the solution.  In my case, I'm in an educational environment where it isn't practical to give hundreds of potential users of the software access to regular docker.  OpenLane works with "rootless docker", but after comparing rootless docker to podman, I feel that the no daemon approach to podman fits the environment better.

PR #1205 (2022-07-18): Don't require `click` for Docker platform extraction
~ `docker/utils.py current-docker-platform` isolated into standalone `docker/current_platform.py` script that does not require any non-default Python libraries

---
Resolves #1200.
closed 2022-07-18 by donn