The-OpenROAD-Project/OpenLane #1206: Max fanout Limit ?

AdhithyanHaridas, 2022-07-14:
Hello,
My design had max fanout violations. The following image shows default fanout limit is set to 5
[image]

I increased the value of the variable "SYNTH_MAX_FANOUT" to 100 in my config and got no fanout violations. 
[image]

Is it okay to exceed the default fanout limit to get timing clean ? or any suggestions ? 

Thanks

maliberty, 2022-07-14:
At what step in the flow is this report from?  repair_design should fix these.

AdhithyanHaridas, 2022-07-14:
The report is from signoff.
FYI, 
The first image is from a run which completed the flow with warnings for max slew violations.

The second image is from a run which completed with no warnings. (SYNTH_MAX_FANOUT set to100)

maliberty, 2022-07-14:
If you look at the net of _42984_/Y what are the fanouts?  I'm wondering if they are something specific like antenna cells or are real fanouts.

antonblanchard, 2022-07-14:
Or perhaps the resizer has been turned off. @AdhithyanHaridas can you attach your config?

vvvverre, 2022-07-18:
I found this thread after encountering a similar issue. I am new to ASIC design, having previously only done FPGA development, so I am not sure how to reduce the high fanout on these pins. Should it be done in my HDL code, through configuring OpenLane, or another way entirely?

This is the output I got in `reports/signoff/30-rcx_sta.slew.rpt`:
```
===========================================================================
 report_check_types -max_slew -max_cap -max_fanout -violators
============================================================================
max fanout

Pin                                   Limit Fanout  Slack
---------------------------------------------------------
clkbuf_4_3_0_wb_clk_i/X                   5     24    -19 (VIOLATED)
clkbuf_4_0_0_wb_clk_i/X                   5     22    -17 (VIOLATED)
clkbuf_4_1_0_wb_clk_i/X                   5     20    -15 (VIOLATED)
clkbuf_4_2_0_wb_clk_i/X                   5     20    -15 (VIOLATED)
clkbuf_4_6_0_wb_clk_i/X                   5     20    -15 (VIOLATED)
clkbuf_4_12_0_wb_clk_i/X                  5     16    -11 (VIOLATED)
clkbuf_4_5_0_wb_clk_i/X                   5     14     -9 (VIOLATED)
clkbuf_4_7_0_wb_clk_i/X                   5     14     -9 (VIOLATED)
clkbuf_4_8_0_wb_clk_i/X                   5     14     -9 (VIOLATED)
clkbuf_4_9_0_wb_clk_i/X                   5     14     -9 (VIOLATED)
clkbuf_4_10_0_wb_clk_i/X                  5     12     -7 (VIOLATED)
clkbuf_4_13_0_wb_clk_i/X                  5     12     -7 (VIOLATED)

<SNIP>

===========================================================================
max slew violation count 0
max fanout violation count 293
max cap violation count 0
============================================================================
````

Based on this thread I tried to increase SYNTH_MAX_FANOUT to 100, and got far fewer violations:
```
===========================================================================
 report_check_types -max_slew -max_cap -max_fanout -violators
============================================================================
max fanout

Pin                                   Limit Fanout  Slack
---------------------------------------------------------
_3629_/X                                100    116    -16 (VIOLATED)
_3759_/X                                100    116    -16 (VIOLATED)
_3589_/X                                100    104     -4 (VIOLATED)

===========================================================================
max slew violation count 0
max fanout violation count 3
max cap violation count 0
============================================================================
```

I have attached the `config.tcl` from the runs directory below:
[openlane_config.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/9133760/openlane_config.txt)

> If you look at the net of 42984/Y what are the fanouts? I'm wondering if they are something specific like antenna cells or are real fanouts.

How would I be able to check this?

I would appreciate any advice or guidance you might be able to give me.

maliberty, 2022-07-18:
re: How would I be able to check this?
I would open the design in the GUI and select the net by name.  The inspector will show you the fanouts. 
You could script it in tcl instead but that would require more knowledge.

SYNTH_MAX_FANOUT = 100 is a bad idea.  You got fewer violations but they are much worse.

maliberty, 2022-07-18:
This is up to the OL team to set @donn @kareefardi but I think the current value of 5 is far too low.  max cap/slew are much more useful in general.

maliberty, 2022-07-18:
I see it is set in [base.sdc](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/base.sdc) to
```
set_max_fanout $::env(SYNTH_MAX_FANOUT) [current_design]
```
so it is really a control for synthesis.  OL carries that through the whole flow and it shows up at the end which I think is a bug in the flow.  There is really no need for it at the end of the flow as max cap & slew are more useful at that point.

It is likely ok to leave at 5 at synthesis time (though I'm not sure if it is beneficial or not).   I guess my opinion is clear but I don't control OL's checks.  I don't see anything to fix in OR.

maliberty, 2022-07-18:
Counting the antenna diodes as fanout effectively reduces max fanout to 2.5 which is even more absurd.

antonblanchard, 2022-07-18:
@vvvverre as @maliberty says, you want to pay more attention to the cap/slew violations. It looks like your fan out violations are in clock tree synthesis (which pays no attention to max fan out constraints anyway) and antenna fix up, which is done after the resizer fixes fan outs. Unless the fan out issues are causing cap/slew violations, I'd ignore it.

As far as a setting for SYNTH_MAX_FANOUT, the current value (5) is almost certainly too low, but due to various issues 100 will be problematic too. I'm trying to get the Openlane default bumped to 10 as an interim improvement.

closed 2022-07-19 by donn

vvvverre, 2022-07-19:
Thanks everyone for the helpful answers. I'll start paying more attention to max cap / slew.

PR #1187 (2022-07-19): Increase SYNTH_MAX_FANOUT from 5 to 10
~ Default value of `SYNTH_MAX_FANOUT` changed from 5 to 10

---

Resolves #1206