The-OpenROAD-Project/OpenLane #1213: absolute pdn offset

kareefardi, 2022-07-18:
### Prompt
A clear and concise description of what shortcoming you feel OpenLane has.

PDN's offset is measured from the core origin. This is the way openroad api is defined. During developing a design core margin can change frequently resulting in a new PDN interface the given same PDN offset

### Proposal
A clear and concise description of what you want to happen.

Using the core margin, core site dimensions, a fixed PDN offset, an offset relative to the core can be calculated at run time to maintain a "fixed" PDN offset. This can ease development of designs and maintenance of new versions of designs

maliberty, 2022-07-18:
Why would you want PDN to be fixed if the core area is moving underneath it?  You may wind up with a poor grid that doesn't cover the core area.

maliberty, 2022-07-18:
It would connect to the straps only over the overlapping area (in the extreme they would be unconnected if there was no overlap).  What is the value in having the PDN fixed in space?

kareefardi, 2022-07-18:
In some cases a given macro is already integrated and it would be hard to change the routing of the parent design

maliberty, 2022-07-18:
Why change the core margin in that case?

maliberty, 2022-07-18:
Do you mean make the margin smaller?  Increasing the margin would increase congestion.  I'm not sure why you don't just set it to a minimal value to start with.

kareefardi, 2022-07-18:
Yes that's what I meant. If the old was big, decreasing it would help. Having a minimal value.
I don't know really know on the top of my head why we aren't using always a minimal margin. I think the relation between the margin, number of pins, congestion, etc.. isn't always directly correlated. I recall that I once had to increase the margin in order to get it to cleanly route.

maliberty, 2022-07-18:
>  I recall that I once had to increase the margin in order to get it to cleanly route.
That sounds more like a bug.  I think using a minimal value would be better than fixing the pdn grid.