The-OpenROAD-Project/OpenLane #1214: make failed

RanChenRivai, 2022-07-19:
When I follow the instructions to execute
    cd OpenLane/
    make
 ,the following error is reported, how should I fix it? Thanks

------------------------------------------------------------------------------------------------------------------------
chenran@rivai-u20-vm-intern:~/work_dir/OpenLane$ make
make[1]: Entering directory '/work/home/chenran/work_dir/OpenLane'
ab6b79bce7f928a9cec34ba672478398a83639e9: Pulling from efabless/openlane
Digest: sha256:36c2fb291a6c0173b637172f0429c44558d4deed3fd744b58247097f70901e60
Status: Image is up to date for efabless/openlane:ab6b79bce7f928a9cec34ba672478398a83639e9
docker.io/efabless/openlane:ab6b79bce7f928a9cec34ba672478398a83639e9
make[1]: Leaving directory '/work/home/chenran/work_dir/OpenLane'
./venv/bin/python3 -m pip install --upgrade --no-cache-dir volare
Requirement already up-to-date: volare in ./venv/lib/python3.8/site-packages (0.2.1)
Requirement already satisfied, skipping upgrade: pcpp~=1.30 in ./venv/lib/python3.8/site-packages (from volare) (1.30)
Requirement already satisfied, skipping upgrade: click<9,>=8.0.0 in ./venv/lib/python3.8/site-packages (from volare) (8.1.3)
Requirement already satisfied, skipping upgrade: rich~=10.16.0 in ./venv/lib/python3.8/site-packages (from volare) (10.16.2)
Requirement already satisfied, skipping upgrade: pyyaml~=5.4.0 in ./venv/lib/python3.8/site-packages (from volare) (5.4.1)
Requirement already satisfied, skipping upgrade: requests~=2.27.0 in ./venv/lib/python3.8/site-packages (from volare) (2.27.1)
Requirement already satisfied, skipping upgrade: click-default-group~=1.2.2 in ./venv/lib/python3.8/site-packages (from volare) (1.2.2)
Requirement already satisfied, skipping upgrade: commonmark<0.10.0,>=0.9.0 in ./venv/lib/python3.8/site-packages (from rich~=10.16.0->volare) (0.9.1)
Requirement already satisfied, skipping upgrade: colorama<0.5.0,>=0.4.0 in ./venv/lib/python3.8/site-packages (from rich~=10.16.0->volare) (0.4.5)
Requirement already satisfied, skipping upgrade: pygments<3.0.0,>=2.6.0 in ./venv/lib/python3.8/site-packages (from rich~=10.16.0->volare) (2.12.0)
Requirement already satisfied, skipping upgrade: urllib3<1.27,>=1.21.1 in ./venv/lib/python3.8/site-packages (from requests~=2.27.0->volare) (1.26.10)
Requirement already satisfied, skipping upgrade: charset-normalizer~=2.0.0; python_version >= "3" in ./venv/lib/python3.8/site-packages (from requests~=2.27.0->volare) (2.0.12)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in ./venv/lib/python3.8/site-packages (from requests~=2.27.0->volare) (2022.6.15)
Requirement already satisfied, skipping upgrade: idna<4,>=2.5; python_version >= "3" in ./venv/lib/python3.8/site-packages (from requests~=2.27.0->volare) (3.3)
./venv/bin/volare enable
[14:25:14] Found version 41c0908b47130d5675ff8484255b43f66463a7d6 in ./dependencies/tool_metadata.yml.                                               common.py:131
Traceback (most recent call last):
  File "/usr/lib/python3.8/pathlib.py", line 1288, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/work/home/chenran/work_dir/OpenLane/pdks/volare/sky130'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./venv/bin/volare", line 8, in <module>
    sys.exit(cli())
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/volare/manage.py", line 255, in enable_cmd
    enable(pdk_root=pdk_root, pdk=pdk, version=version)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/volare/manage.py", line 148, in enable
    mkdirp(current_file_dir)
  File "/work/home/chenran/work_dir/OpenLane/venv/lib/python3.8/site-packages/volare/common.py", line 44, in mkdirp
    return pathlib.Path(path).mkdir(parents=True, exist_ok=True)
  File "/usr/lib/python3.8/pathlib.py", line 1292, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "/usr/lib/python3.8/pathlib.py", line 1288, in mkdir
    self._accessor.mkdir(self, mode)
PermissionError: [Errno 13] Permission denied: '/work/home/chenran/work_dir/OpenLane/pdks/volare'
make: *** [Makefile:107: pdk] Error 1

RanChenRivai, 2022-07-19:
> 

Yes, the problem is that the makefile doesn't create anything in /OpenLane/pdks/
--------------------------------------------------------------------------------------------------------------------------
chenran@rivai-u20-vm-intern:~/work_dir/OpenLane$ cd /work/home/chenran/work_dir/OpenLane/pdks/
chenran@rivai-u20-vm-intern:~/work_dir/OpenLane/pdks$ ll
total 8
drwxr-xr-x  2 root    root      4096 Jul 19 10:29 ./
drwxr-xr-x 15 chenran engineers 4096 Jul 19 11:55 ../

vijayank88, 2022-07-19:
@RanChenRivai 
try `sudo make`

vijayank88, 2022-07-19:
@RanChenRivai 
Once `make` completes without any issues, you may close this issue.

donn, 2022-07-19:
Issue template is missing.

closed 2022-07-19 by donn