The-OpenROAD-Project/OpenLane #1218: Can I run an ECO flow even if there aren't any timing violations?

jarvis-lorry, 2022-07-19:
Is it possible run the complete OL flow, change something in the LEF or DEF files and then have the tool rerun the route, keeping everything placed the same?  I don't have any timing violations, but I'm trying to see how changes to the library affect routing.  Does the current ECO flow allow for this?

vijayank88, 2022-07-20:
Use interactive mode. So that you can use updated file as `CURRENT_DEF`

donn, 2022-07-21:
@jarvis-lorry As @vijayank88 stated, you can use interactive mode (`./flow.tcl -it`) which allows you to run functions from the flow step-by-step, but you can also try `./flow.tcl -last_run -from routing` after your modifications, for example.

jarvis-lorry, 2022-07-21:
Thank you, @vijayank88 and @donn.  @vijayank88, using the interactive flow does work for me but occasionally, OL errors out, probably from a mistake I've made.  @donn, I will try your suggestion today.

jarvis-lorry, 2022-07-21:
> @jarvis-lorry As @vijayank88 stated, you can use interactive mode (`./flow.tcl -it`) which allows you to run functions from the flow step-by-step, but you can also try `./flow.tcl -last_run -from routing` after your modifications, for example.

@donn , if I use your second option, how do I set the CURRENT_LEF and MERGED_LEF_UNPADDED?

donn, 2022-07-21:
`CURRENT_DEF`, you mean? There's a `config.tcl` file in the run. You can modify the last `CURRENT_DEF` in it, same for `MERGED_LEF_UNPADDED`.

You can also try passing `-override_env CURRENT_DEF=path_to_x,MERGED_LEF_UNPADDED=path_to_y` but I must confess I haven't tested that.

jarvis-lorry, 2022-07-25:
@donn I've tried to use your suggestion but I'm having problems.  I can run './flow -design spm -last_run -from routing' but I cannot change the CURRENT_DEF because no change I make to config.tcl is reflected in the run.  In the run directory are two config files, config.tcl and config_in.tcl, and these are overwritten by OpenLane after each run.  Even if I change the value of CURRENT_DEF in config.tcl, it's ignored and overwritten.  OpenLane now uses config.json in the design directory when it starts compiling.  I've modified config.json but still I can't change the value of CURRENT_DEF.

PR #1685 (2023-03-21): Exit Cleanup
~ Move `QUIT_ON` variables outside of checker functions
~ Move final timing checks to a flow step: `run_timing_check_step`
~ Replace a couple of loose `file exists` with `assert_files_exist` calls
~ Replace all `flow_fail` across the flow with `throw_error`, which behaves accordingly:
    * If running an interactive script, `flow_fail` is called
    * Else, the errors are propagated upwards, where `flow.tcl` is to catch it
~ Rename `try_catch` to `try_exec` as it now propagates the error, with translation behavior
- Remove UMich experimental ECO flow
---
Resolves #1079
Resolves #1 [... cut]
closed 2023-03-21 by donn