The-OpenROAD-Project/OpenLane #1224: Update OpenROAD app to get JSON metrics features

vvbandeira, 2022-07-20:
### Prompt

OpenROAD now has improved support for JSON metrics.

### Proposal

Update OpenROAD app to the latest version to pick up code changes to support JSON metrics.
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2047

donn, 2022-07-21:
I can assure you assigning me things won't get them done any faster

vijayank88, 2022-07-21:
@donn 
Is there any automated PR to pick latest OpenROAD commit and update OpenLane automatically?

donn, 2022-07-21:
https://github.com/The-OpenROAD-Project/OpenLane/pull/1219

donn, 2022-07-21:
However, I'm not going to merge automatic OpenROAD updates if things are broken, which they are.

maliberty, 2022-07-26:
@donn what is the next action here?  I don't see any failures in #1219 except a build timeout

donn, 2022-07-26:
@maliberty I need to build OR for ARM and upload it

maliberty, 2022-07-26:
@donn You unassigned yourself so I wasn't clear who is taking ownership at this point.  Do you require anything from OR?

donn, 2022-07-26:
Nope.

PR #1219 (2022-07-26): [BOT] Update openroad_app
This is an automated PR.

See the individual commits for details.

---
Resolves #1224
closed 2022-07-26 by donn