The-OpenROAD-Project/OpenLane #1227: Cannot rerun 'run_routing' in interactive flow

jarvis-lorry, 2022-07-21:
With the latest version of OpenLane, I cannot run routing more than once in interactive mode because it errors out.

ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: designs/spm/runs/run3/logs/routing/27-resizer.log
[ERROR]: Last 10 lines:
met5               208             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total             7098             0            0.00%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_timing.tcl, 59 RSZ-0005
child process exited abnormally

[ERROR]: Creating issue reproducible...

I'm including the error log here because right now GitHub won't accept any uploaded files.

vijayank88, 2022-07-22:
share the command what you tried in interactive mode

donn, 2022-07-22:
Issue template is invalid.

closed 2022-07-22 by donn

jarvis-lorry, 2022-07-22:
I've created a new bug report using the required template:  With latest version of OpenLane, I cannot run routing more than once in interactive mode #1230.

@vijayank88, thank you for you response.  I've answered your questions in the new bug report I've created.