The-OpenROAD-Project/OpenLane #1230: With latest version of OpenLane, I cannot run routing more than once in interactive mode

jarvis-lorry, 2022-07-22:
### Description
With the latest version of OpenLane, I cannot run routing more than once in interactive mode because it errors out.  I'm running the example design spm as a test case.

### Environment
```
make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 23ab827baca7eefc0ca367d2e624241a3bbd95a5
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

23ab827 2022-07-20T14:57:39+02:00 GLB_RT -> GRT + Remove older deprecated variables (#1217) - Mohamed Gaber -  (HEAD -> master, tag: 2022.07.21, origin/master, origin/HEAD)
2f94230 2022-07-20T14:53:36+02:00 New Tagging Scheme + Issue Features (#1216) - Mohamed Gaber -  ()
b52022f 2022-07-19T18:53:54+02:00 Increase SYNTH_MAX_FANOUT from 5 to 10 (#1187) - Anton Blanchard -  (tag: 2022.07.20_01.44.12)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.
./flow.tcl -design spm -tag run3 -overwrite -interactive
run_synthesis
run_floorplan
run_placement
run_cts
run_routing
run_routing

[run_routing_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9168926/run_routing_issue_reproducible.tar.gz)
### Expected behavior
I would like to run the complete flow in interactive mode, then modify the DEF or LEF, then rerun the routing to see how changing the DEF/LEF affects the final output.

### Logs
```
ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: designs/spm/runs/run3/logs/routing/27-resizer.log
[ERROR]: Last 10 lines:
met5 208 0 0.00% 0 / 0 / 0

Total 7098 0 0.00% 0 / 0 / 0

[INFO GRT-0018] Total wirelength: 0 um
[INFO GRT-0014] Routed nets: 0
[INFO]: Setting RC values...
[ERROR RSZ-0005] Run global_route before estimating parasitics for global routing.
Error: resizer_routing_timing.tcl, 59 RSZ-0005
child process exited abnormally

[ERROR]: Creating issue reproducible...
[27-resizer.log.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9168930/27-resizer.log.log)
```

donn, 2022-07-22:
Thanks- we'll investigate.

thesourcerer8, 2023-02-26:
I am getting the same error message with a non-interactive run sporadically, you can get the reports and all files here: https://github.com/thesourcerer8/gf180teststructure/actions/runs/4275011066

donn, 2024-07-23:
To have the ability to repeat steps, try OpenLane 2: https://github.com/efabless/openlane2

closed 2024-07-23 by donn