The-OpenROAD-Project/OpenLane #1231: How to fix antenna violations ?

AdhithyanHaridas, 2022-07-24:
My design has Antenna violations, I tried the following
1. Increasing the target density
2. Used different floorplan
3. Tried different DIODE_INSERTION_STRATEGY
4. Incresed GRT_MAX_DIODE_INS_ITERS
5. Increased GRT_ANT_ITERS

After trying all the above. I was able to reduce the no. of antenna violations down to 112.
But not able to fix it completely. 

[image]

Any suggestions ?

Sorry, I wouldnt be able to share the repo here because, the project is closed source for chipignite tape out.

AdhithyanHaridas, 2022-07-28:
On the Skywater slack, @RTimothyEdwards mentioned that most violations are not worth fixing unless they exceed the ratio by 2x.

Considering that, out of 117 violated nets, only 42 nets are valid (more than 2x).

_Below is the snapshot from Antenna report, where net 55 exceeds the 2x ratio._
[image]

_Below is the snapshot from OpenRoad gui, net 55 has antenna connected to it._
[image]

I checked all the 42 Violating nets, and they have antenna cells connnected.

Although all violating nets have antenna cells connected, why are they still reported as violations ?

@maliberty @antonblanchard @vijayank88

maliberty, 2022-07-28:
It depends where the diode is placed.  You could have an issue in another part of the net that isn't connected to this diode when the violation occurs.

maliberty, 2022-07-28:
It would be good to include a test case

closed 2024-07-17 by kareefardi