The-OpenROAD-Project/OpenLane #1232: difference between sky130A and sky130B

pengpeng-lian, 2022-07-25:
### Proposal
Dear Worker，I know that the PDK library includes Standard Cell Libraries, but I don't know the difference between sky130A and sky130B?
<html>
<body>
<div data-page-id="doxcn7W1GKEQDnRY3l1XPgzpHZd" data-docx-has-block-data="true"><div data-type="sheet" class=" old-record-id-doxcny6KwCACqs8omEfNGt9nwHh"><style> td {white-space:pre-wrap;border:1px solid #dee0e3;}</style><byte-sheet-html-origin data-id="hJYRprQlBa-1658742416761" data-version="3" data-is-embed="true">

  | :lib:`sky130_fd_sc_lp` | :lib:`sky130_fd_sc_ls` | :lib:`sky130_fd_sc_ms` | :lib:`sky130_fd_sc_hs` | :lib:`sky130_fd_sc_hd` | :lib:`sky130_fd_sc_hdll` | :lib:`sky130_fd_sc_hvl`
-- | -- | -- | -- | -- | -- | -- | --
VDD | 1.8 | 1.8 | 1.8 | 1.8 | 1.8 | 1.8 | 5
NMOS devices used | :cell:`sky130_fd_pr__nfet_01v8` | :cell:`sky130_fd_pr__nfet_01v8` | :cell:`sky130_fd_pr__nfet_01v8_lvt` | :cell:`sky130_fd_pr__nfet_01v8_lvt` | :cell:`sky130_fd_pr__nfet_01v8` | :cell:`sky130_fd_pr__nfet_01v8` | :cell:`sky130_fd_pr__nfet_g5v0d10v5`
PMOS devices used | :cell:`sky130_fd_pr__pfet_01v8_hvt` | :cell:`sky130_fd_pr__pfet_01v8_hvt` | :cell:`sky130_fd_pr__pfet_01v8` | :cell:`sky130_fd_pr__pfet_01v8_lvt` | :cell:`sky130_fd_pr__pfet_01v8` | :cell:`sky130_fd_pr__pfet_01v8_hvt` | :cell:`sky130_fd_pr__pfet_g5v0d10v5`
inverters, buffers | 108 | 48 | 48 | 48 | 56 | 62 | 19
AND, OR, NAND, NOR | 159 | 66 | 86 | 86 | 153 | 170 | 8
XOR, XNOR | 16 | 12 | 12 | 12 | 8 | 10 | 2
AND-OR-INV, OR-AND-INV | 138 | 71 | 71 | 71 | 115 | 125 | 4
AND-OR, OR-AND | 132 | 68 | 68 | 68 | 132 | 134 | 4
Adders, Comparators, Multiplexors | 59 | 37 | 33 | 33 | 31 | 44 | 11
Latches and Flip-Flops | 92 | 68 | 68 | 68 | 60 | 60 | 17
Custom power gating, bus cells | 43 | 66 | 42 | 42 | 51 | 42 |  

</byte-sheet-html-origin></div></div><span data-lark-record-data="{&quot;isCut&quot;:false,&quot;rootId&quot;:&quot;doxcn7W1GKEQDnRY3l1XPgzpHZd&quot;,&quot;parentId&quot;:&quot;doxcn7W1GKEQDnRY3l1XPgzpHZd&quot;,&quot;blockIds&quot;:[47],&quot;recordIds&quot;:[&quot;doxcny6KwCACqs8omEfNGt9nwHh&quot;],&quot;recordMap&quot;:{&quot;doxcny6KwCACqs8omEfNGt9nwHh&quot;:{&quot;id&quot;:&quot;doxcny6KwCACqs8omEfNGt9nwHh&quot;,&quot;snapshot&quot;:{&quot;author&quot;:&quot;7108543137328398364&quot;,&quot;comments&quot;:[],&quot;hidden&quot;:false,&quot;locked&quot;:false,&quot;parent_id&quot;:&quot;doxcn7W1GKEQDnRY3l1XPgzpHZd&quot;,&quot;token&quot;:&quot;shtcnjZXdmJ3pymagvrpEx9hH2f_JOTrJw&quot;,&quot;type&quot;:&quot;sheet&quot;}},&quot;doxcn7W1GKEQDnRY3l1XPgzpHZd&quot;:{&quot;id&quot;:&quot;doxcn7W1GKEQDnRY3l1XPgzpHZd&quot;,&quot;snapshot&quot;:{&quot;align&quot;:&quot;&quot;,&quot;author&quot;:&quot;7108543137328398364&quot;,&quot;children&quot;:[&quot;doxcnusWww8momSiIyGMyYlt6xe&quot;,&quot;doxcnWISSUwCuOwuItDW9wCP7se&quot;,&quot;doxcnAq0qWYOOaEGY2np6Rgq5gf&quot;,&quot;doxcn2M8ssaEmOsm0svqpJWdP5g&quot;,&quot;doxcnwIuSoeC8YeKeEVWiK4y2pd&quot;,&quot;doxcn02igecc2sacusJPxRjr8fd&quot;,&quot;doxcn0YAEgaEaEiSEmQ6SZKydFh&quot;,&quot;doxcnGUmM2iMgQiWg63JrvYTH9d&quot;,&quot;doxcn6M2Y2ouSeuiyaortKKLnSh&quot;,&quot;doxcnS4weSuwuGo20qU1v4SiQ9f&quot;,&quot;doxcnaiWU6IqiKSoqgdwcy2xp5e&quot;,&quot;doxcnsiQKUOcYKECKykHLPB5mrh&quot;,&quot;doxcnGYI8CmieAesy2HBSfC0L3e&quot;,&quot;doxcnOkW0eCa00csSiW4MG080id&quot;,&quot;doxcnU08eEyCC626csTObnSDWhe&quot;,&quot;doxcnGMGiMOSIOyQoScspZykT69&quot;,&quot;doxcnw0mYEMmOQA224MNKhDPGRc&quot;,&quot;doxcni860uGye40sIJltlSdOrlN&quot;,&quot;doxcnc2MgiysWskusWEvCI3opDd&quot;,&quot;doxcn6MeaGWqGIEgqcnY5mtqhDe&quot;,&quot;doxcn0MwimYmgESkIGmkCMAjX3e&quot;,&quot;doxcnoEsYAem8AecWc5ovKSuGRf&quot;,&quot;doxcn662muoKGWgQaMb9halTgeg&quot;,&quot;doxcnyEs2GQiCe2SIugscBa3RZb&quot;,&quot;doxcnkMsG0OQuGI8aqcVbQl8Nme&quot;,&quot;doxcnqCgiOEISuIgoWKooVONPLh&quot;,&quot;doxcnsqis46oscswamyZbKJGf5d&quot;,&quot;doxcncMwAsOYEIUceOAzPHH1zQd&quot;,&quot;doxcn2GOKm6Y6AgkCi8qo1HXqUd&quot;,&quot;doxcnmsUGYU4c4oM0ID3jwuvfPb&quot;,&quot;doxcnuosYg8W8iMwEWauYWMbh7e&quot;,&quot;doxcngcEuiEEqUIGu4OcpT8yHrd&quot;,&quot;doxcn8sWWAICO6KEwGKghJ4AQmf&quot;,&quot;doxcnKmyUMYuUCuMqmgmc7CdrBh&quot;,&quot;doxcn6oc2OYo0S0w6k3prNAAUSf&quot;,&quot;doxcng6UCYA0uGmSEGCGVuIUJZc&quot;,&quot;doxcny6KwCACqs8omEfNGt9nwHh&quot;,&quot;doxcnaysoWuSUOGGe6L0eKXJr5c&quot;,&quot;doxcnigMoKUeggWqgwrXVMYzTqe&quot;,&quot;doxcnSQiWIuOK6oSWMRGGh5VCug&quot;,&quot;doxcn6u6mYMwwECIQc7zPFzKtYe&quot;,&quot;doxcnQu4wSq0G4UoYUlGymLm8Tc&quot;,&quot;doxcn0g0UcgeqYEQ6ATLBWeGSLc&quot;],&quot;comments&quot;:null,&quot;hidden&quot;:false,&quot;locked&quot;:false,&quot;parent_id&quot;:&quot;&quot;,&quot;text&quot;:{&quot;apool&quot;:{&quot;nextNum&quot;:1,&quot;numToAttrib&quot;:{&quot;0&quot;:[&quot;author&quot;,&quot;7108543137328398364&quot;]}},&quot;initialAttributedTexts&quot;:{&quot;attribs&quot;:{&quot;0&quot;:&quot;*0+7&quot;},&quot;text&quot;:{&quot;0&quot;:&quot;PDK库的信息&quot;}}},&quot;type&quot;:&quot;page&quot;}}},&quot;payloadMap&quot;:{},&quot;extra&quot;:{&quot;mention_page_title&quot;:{},&quot;external_mention_url&quot;:{}},&quot;pasteFlag&quot;:&quot;82b7d8ad-8497-466f-90bb-2c25b903d9ce&quot;}" data-lark-record-format="docx/record" class="lark-record-clipboard"></span>
</body>
</html>

donn, 2022-07-25:
For more context, the OpenLane team refers to "sky130A" and "sky130B" as **variants** of the sky130 PDK. It's the same PDK, just with different optional modules included.

closed 2022-07-25 by donn

donn, 2022-07-25:
Was just thinking that.

pengpeng-lian, 2022-07-26:
> Was just thinking that.

Thank you very much for your answer, it was very useful to me and I hope to work with you next time! Would be great if there was some documentation somewhere that you could link to for questions like this.