The-OpenROAD-Project/OpenLane #1236: Incorporate JSON metrics report from OpenROAD into OpenLane

ravi-varadarajan, 2022-07-26:
OpenROAD currently dumps of metrics in JSON format with the use of -metrics <metrics_file_name>. There are also dedicated tcl commands to dump out the area/timing/power metrics  which can called during the various stages of the flow.

We would like to modify the OpenLane scripts to use the -metrics option when invoking OpenROAD and also update the tcl scripts to set the design stage and invoke the appropriate metrics reporting functions. Once the final stage is done, we will need a utility function like genMetrics.py in ORFS  to aggregate all of the metrics.

I can help in updating the flow

dralabeing, 2022-07-27:
@ravi-varadarajan Thanks. Is there a document to refer to the tcl command usage?. Is this a general metrics reporting api- i.e can users independently invoke these scripts to generate/customize metrics reporting  outside of OpenLane for example?

donn, 2023-04-13:
Tracked in #1607.

closed 2023-04-13 by donn