The-OpenROAD-Project/OpenLane #1238: Increasing SYNTH_MAX_FANOUT results in crash during PDN

mattvenn, 2022-07-27:
### Description

I was increasing SYNTH_MAX_FANOUT to try to remove the warning about fanout. At 15, the PDN fails and the flow stops with an error.

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

### Reproduction Material

* Upload a tarball containing the relevant design. Have done. Didn't bother with issue_reproducible - see below. 

[RUN_2022.07.27_14.37.12.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9200169/RUN_2022.07.27_14.37.12.tar.gz)

./flow -design seven_segment_seconds

### Expected behavior

Tools should finish the flow.

### Other questions

Why is issue_reproducible so huge? The uncompressed directory is 7.2GB. I had a quick look and it seems to recursively pack itself.
28k files. Compressed with tar.gz -> 1.2GB. Didn't bother attaching here.

maliberty, 2022-07-27:
@donn another instance of recursion blocking a reproducer.

maliberty, 2022-07-27:
What was the pdn error?

mattvenn, 2022-07-27:
```
[STEP 6]                                                                                                                                                  
[INFO]: Generating PDN...                                                                                                                                 
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl                                                                              
[ERROR]: Exit code: 1                                                                                                                                     
[ERROR]: full log: designs/seven_segment_seconds/runs/RUN_2022.07.27_14.37.12/logs/floorplan/6-pdn.log                                                    
[ERROR]: Last 10 lines:                                                                                                                                   
[INFO ODB-0226] Finished LEF file:  /openlane/designs/seven_segment_seconds/runs/RUN_2022.07.27_14.37.12/tmp/merged.unpadded.nom.lef                      
[INFO ODB-0127] Reading DEF file: /openlane/designs/seven_segment_seconds/runs/RUN_2022.07.27_14.37.12/results/floorplan/seven_segment_seconds.def        
[INFO ODB-0128] Design: seven_segment_seconds                                                                                                             
[INFO ODB-0130]     Created 34 pins.                                                                                                                      
[INFO ODB-0131]     Created 703 components and 3735 component-terminals.                                                                                  
[INFO ODB-0133]     Created 422 nets and 1355 connections.                                                                                                
[INFO ODB-0134] Finished DEF file: /openlane/designs/seven_segment_seconds/runs/RUN_2022.07.27_14.37.12/results/floorplan/seven_segment_seconds.def
[ERROR PDN-0191] Offset of 15.6980 does not fit the manufacturing grid of 0.0050.
Error: pdn_cfg.tcl, 127 PDN-0191
child process exited abnormally

```

mattvenn, 2022-07-27:
very strange that synth_max_fanout would cause this

maliberty, 2022-07-27:
You probably get a smaller design and then hit the known hack that is FP_PDN_AUTO_ADJUST.  Did you get the message "Current core area is too small for a power grid. The power grid will be minimized."

donn, 2022-07-27:
@maliberty Recursion was fixed in 281281cc94c2084302845683f2e33fae6cf54eaa

I'll take a look at snapping these to a grid- I ran into this recently myself

PR #1255 (2022-08-09): IR Drop Reports in the Signoff Stage
\+ Create new IR drop report generation step in the signoff stage
\+ ALL* STEPS NOW PRINT THEIR LOGS' PATH
~ `increment_index/TIMER/set log/puts_info` commands organized across the board
~ Fix issue where `FP_PDN_AUTO_ADJUST` would not snap its values to the mfg grid
~ Reorder `configuration/general.tcl`
~ Update Contribution Guidelines
~ Various Docs Fixes
\- Remove `FP_PDN_IRDROP`

\* All /ɔːl/ *noun*: used to refer to the whole quantity or extent of a particular group or thing so long as said thing does not include global routing or lvs as they are writt [... cut]
closed 2022-08-09 by donn