The-OpenROAD-Project/OpenLane #1243: [ERROR GRT-0118] Routing congestion too high with latest Openlane Tool (Git version: 4476a58407d670d251aa0be6a55e5391bb181c4e) Same design was sucessfully routed with older version of openlane tool set.

dineshannayya, 2022-07-31:
### Description
I have been using this macro being used from MPW-3 onwards. This block was successfully rooted in previous version of Openlane tool tools set from MPW-3 to MPW-6.  But with latest Openlane update I am unable to route this block.

I am looking suggestion to Route this block.  This block has lot's of interconnect signal and PL_TARGET_DENSITY is set a small value of 0.20. 
  
### Environment

<pre>python3 ./env.py issue-survey
Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 4476a58407d670d251aa0be6a55e5391bb181c4e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (HEAD -&gt; master)
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  ()
0b55101 2022-07-26T20:37:20+02:00 [BOT] Update openroad_app (#1219) - Openlane Bot -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

### Reproduction Material
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9229151/issue_reproducible.tar.gz)

### Expected behavior
Clean Openlane flow

### Logs
<pre>ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Log: wb_interconnect/runs/wb_interconnect/logs/routing/20-global.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 56774
[INFO GRT-0198] Via related Steiner nodes: 1329
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 94671
[INFO GRT-0112] Final usage 3D: 485592
[ERROR GRT-0118] Routing congestion too high.
Error: groute.tcl, 55 GRT-0118
child process exited abnormally
</pre>

donn, 2022-07-31:
@maliberty Thoughts?

maliberty, 2022-07-31:
Do you have an example of a run that previously worked?  It could be all sorts of things.  All I can say now is that you have lots of vertical routing congestion that seems real.

dineshannayya, 2022-08-01:
@maliberty  This block was successfully routed from MPW-3 to MPW6 tool set.  Normally this configuration set ::env(CELL_PAD) "8" use help in previous version of the tool.

Attached the working case with MPW6 tool set.  I have created this at  scripts/openroad/droute.tcl  stage.

[working.mpw6.openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9231065/working.mpw6.openroad_issue_reproducible.tar.gz)

<pre>Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -&gt; master, tag: 2022.05.18_02.12.32, origin/master, origin/HEAD)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()
</pre>

vijayank88, 2022-08-01:
@dineshannayya 
If any of previous MPW designs failing please check this : https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI 
Search for respective design name `wb_interconnect` and you can find `scripts/wb_interconnect/config.tcl` : https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/blob/c72552a358466ecb13309ed26ec5993f57aba473/scripts/wb_interconnect/config.tcl is your latest configuration setting to pass `OpenLane` flow.

Normally we've use `RT_MAX_LAYER` as `met5`, for hardening just use as `met4`.

vijayank88, 2022-08-01:
This issue already part of OR issue and resolved: https://github.com/The-OpenROAD-Project/OpenROAD/issues/1655

dineshannayya, 2022-08-01:
@vijayank88 Yes, with addition below two setting I am able complete the routing
set ::env(GRT_ADJUSTMENT) 0.1
set ::env(DPL_CELL_PADDING) 1
We can close this issue

closed 2022-08-01 by dineshannayya