The-OpenROAD-Project/OpenLane #1245: Detailed routing problem

MohamedAliYounis, 2022-08-01:
### Description
This Error appears in detailed routing and I don't know how to solve it, the issue_reproducible and config file are attached.

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-43-generic
Distribution: ubuntu 22.04
Python: v3.10.4 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 4476a58407d670d251aa0be6a55e5391bb181c4e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (HEAD -> master, tag: 2022.07.30, origin/master, origin/HEAD)
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  (tag: 2022.07.28)
0b55101 2022-07-26T20:37:20+02:00 [BOT] Update openroad_app (#1219) - Openlane Bot -  (tag: 2022.07.27)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

```

### Reproduction Material
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9235070/issue_reproducible.zip)

[config (1).zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9235071/config.1.zip)

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
Expected to pass routing

### Logs
```
[INFO]: Running Detailed Routing (logging to '../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/logs/routing/24-detailed.log')...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Log: ../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/logs/routing/24-detailed.log
[ERROR]: Last 10 lines:
Error: net net93 initial #node != #rpin
Error: net net94 initial #node != #rpin
Error: net net95 initial #node != #rpin
Error: net net96 initial #node != #rpin
Error: net net97 initial #node != #rpin
Error: net net98 initial #node != #rpin
Error: net net99 initial #node != #rpin
generating net topology...
Error: pin not found in pin2RPins
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/issue_reproducible…
Found $::env(DRT_OPT_ITERS)…
Found $::env(ROUTING_CORES)…
Found $::env(RT_MAX_LAYER)…
Found $::env(RT_MIN_LAYER)…
Found $::env(CURRENT_DEF)…
Found $::env(CURRENT_GUIDE)…
Found $::env(MERGED_LEF)…
Found $::env(SAVE_DEF)…
Found $::env(SAVE_DEF)…
Found $::env(_tmp_drt_file_prefix)…
Found $::env(_tmp_drt_rpt_prefix)…
Done.
[INFO]: Reproducible packaged: Please tarball and upload '../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/issue_reproducible' if you're going to submit an issue.
[INFO]: Saving current set of views in '../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/younis/caravel/caravel_user_project/openlane/core/runs/22_08_01_17_02/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "run_tcl_script" line 89)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/droute.tcl -indexed_log $drt_log"
    (procedure "detailed_routing_tritonroute" line 24)
    invoked from within
"detailed_routing_tritonroute {*}$args"
    (procedure "detailed_routing" line 2)
    invoked from within
"detailed_routing"
    (procedure "run_routing" line 50)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "/openlane/flow.tcl" line 393)
make[1]: *** [Makefile:72: core] Error 1
make[1]: Leaving directory '/home/younis/caravel/caravel_user_project/openlane'
make: *** [Makefile:73: core] Error 2
```

maliberty, 2022-08-01:
You should be loading guides.  The "initial #node != #rpin" comes from code that is not supported and only happens if you don't provide guides to the router and it tries to invent its own.

MohamedAliYounis, 2022-08-01:
@maliberty 
Sorry but what do you mean by guides? Are they variables in the config file? If you have an example I will be thankful.

maliberty, 2022-08-01:
Guides are generated by the global router to provide guidance to the detailed router.  OL should handle this automaticallly so I am wondering if you are deviating from the default flow somehow.

MohamedAliYounis, 2022-08-01:
The version of openroad is old, when I tried to update it this error occurred. any solution?
[image]

maliberty, 2022-08-01:
@donn any ideas?

antonblanchard, 2022-08-02:
You are trying to build your own openlane docker image and you are hitting DNS resolution errors (so a problem with your setup).

Unless you are developing on openlane, you can just use a recent built image such as `efabless/openlane:2022.07.30`

MohamedAliYounis, 2022-08-02:
@antonblanchard 
If I got you, the main issue while using the latest version: 
~~~
OpenLane Git Version: 4476a58407d670d251aa0be6a55e5391bb181c4e
~~~

The problem of using guides appears, so what's your explanation?

vijayank88, 2022-08-02:
@MohamedAliYounis 
Are you using mpw-7a tag?

MohamedAliYounis, 2022-08-02:
@vijayank88 
I was facing error due to sky130B and it is fixed in commit "tag: 2022.07.30"

MohamedAliYounis, 2022-08-02:
I used the toolset of mpw-7a with the latest open-pdk update and the errors are solved.
I will close this issue.

closed 2022-08-02 by MohamedAliYounis