The-OpenROAD-Project/OpenLane #1246: global placement fails on small design - was working with MPW6 tools

mattvenn, 2022-08-02:
### Description

Global router fails with

```
[ERROR]: during executing openroad script /openlane/scripts/openroad/replace.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/wokwi/runs/RUN_2022.08.02_10.47.51/logs/placement/7-global.log
[ERROR]: Last 10 lines:
[INFO GPL-0030] NumBins: 256
[NesterovSolve] Iter: 1 overflow: 0.464794 HPWL: 597399
[INFO GPL-0100] worst slack 5.91e-08
[INFO GPL-0103] Weighted 9 nets.
[NesterovSolve] Snapshot saved at iter = 0
[INFO GPL-0100] worst slack 5.93e-08
[INFO GPL-0103] Weighted 8 nets.
[ERROR GPL-0306] RePlAce diverged at wire/density gradient Sum.
Error: replace.tcl, 91 GPL-0306
child process exited abnormally
```

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-43-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)

```

### Reproduction Material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9241484/issue.tar.gz)

```
make mount
./flow -design wokwi
```

### Expected behavior

Global router shouldn't fail.

vijayank88, 2022-08-02:
@mattvenn 
OpenLane tag is `2022-07-01`. Have you tried with latest tag?

mattvenn, 2022-08-02:
> @mattvenn <https://github.com/mattvenn>
> Please share the repo link. So I can look into that and try with same tag
> to reproduce the issue.

https://github.com/mattvenn/wokwi-verilog-gds-test

Thanks!

vijayank88, 2022-08-02:
@mattvenn 
As per `config.tcl` its missing https://github.com/mattvenn/wokwi-verilog-gds-test/blob/90be7703aae02f5db47f5a6c9f4e77a7079bc19d/src/config.tcl#L18 `$::env(DESIGN_DIR)/wokwi.v `

vijayank88, 2022-08-02:
@mattvenn 
I am able to fix the issue by updating `set ::env(PL_BASIC_PLACEMENT) {1}`

maliberty, 2022-08-02:
@vijayank88 that is a workaround.  We should still fix the issue.

maliberty, 2022-08-02:
When I run this test case with the current openroad I get no error.  The testcase seems rather old as you still have "-verbose_level 1" in your script.

vijayank88, 2022-08-06:
@maliberty 
MPW-7a tag using OpenROAD commit: 0b8b7ae255f8fbbbefa57d443949b84e73eed757
So latest updates not merged with mpw-7a tag.

maliberty, 2022-08-08:
@vijayank88 as there is no error in the latest OR I don't see anything to fix.

closed 2022-08-09 by mattvenn