The-OpenROAD-Project/OpenLane #1248: [Error] Fails on Running Placement Resizer Design Optimizations

AyaseErii, 2022-08-02:
### Description
The flow fails on the running placement resizer design optimizations. I tried to implement a small array 2x2, which could work perfectly. However, when design was scaled just a little, i.e. 4x4, the flow failed at the step 9 Running Placement Resizer Design Optimizations, and showed an error regarding the `resizer.tcl`. Could anyone help me with this issue? I will appreciate it!

### Environment
```
python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v4.15.0-189-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: a9883335a102c9ba1d862a209a72fc8b4c20c785
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a988333 2022-06-18T11:53:39+02:00 Enhancements + Bugfixes to Multi-PDK Support (#1138) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.19_01.37.45)
732c856 2022-06-18T11:22:57+02:00 Fix `skip_basic_prep`, step mechanics (#1143) - Mohamed Gaber -  ()
074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (tag: 2022.06.15_01.35.48)
```

### Reproduction Material
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9245981/openroad_issue_reproducible.tar.gz)

### Expected behavior
The flow should work just like the smaller array design.

### Logs
```
[STEP 9]
[INFO]: Running Placement Resizer Design Optimizations...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/fpga_core/runs/RUN_2022.08.02_21.58.32/logs/placement/9-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  cbx_3__0_
[INFO DPL-0035]  sb_3__0_
[INFO DPL-0035]  cbx_4__1_
[INFO DPL-0035]  cbx_2__0_
[INFO DPL-0035]  cbx_1__4_
[INFO DPL-0035]  cby_4__1_
[INFO DPL-0035]  cbx_4__0_
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 79 DPL-0036
child process exited abnormally
```

maliberty, 2022-08-02:
Why aren't your macros placed before this step by the macro placer?

AyaseErii, 2022-08-02:
Thanks for the reply. I actually did set macro_placement.cfg file for the placement. Not sure what happened. I probably need to fix the macro_placement.cfg file. Could you tell me where I can find this placement result like your screenshot? Thanks!

maliberty, 2022-08-02:
I just opened your test case with "openroad -gui".  I believe you have a similar -gui option to flow.tcl in OL

AyaseErii, 2022-08-03:
I see what you are saying, thanks, Matt!

maliberty, 2022-08-03:
You are welcome.  Can I close this?

AyaseErii, 2022-08-03:
Yes, thanks!

closed 2022-08-03 by maliberty