The-OpenROAD-Project/OpenLane #1249: [Error] XOR was killed with exit code 1. How to set a fixed die area for a small design/macro?

AyaseErii, 2022-08-05:
### Description
I am trying to generate layouts of 3 designs but with different sizes, i.e. small, medium, and large, using macro hardening, i.e. each of the 3 designs contains several macros. For the small design (DIE AREA 1600 1700), it works well to generate the layout. For the medium design, first, I set the DIE AREA as 4000 4500), it failed with the error message shown in the log in the end. Then I tried to shrink the gaps between macros of this medium design to have a relatively smaller total area like DIE AREA 2900 3400, it passed the step and the flow was completed. For the large design, I set the DIE AREA as 5200 5800 after shrinking the gaps between macros, the flow fails at the same step with the same error message shown in the log in the end. Since smaller total area may help with passing the flow, I tried to make the die area of each macro smaller. However, as I set `FP_SIZING "absolute"` and `DIE_AREA "0 0 100 86"` the step 3 shows:
```
[STEP 3]
[INFO]: Running Initial Floorplanning...
[WARNING]: Current core area is too small for a power grid. The power grid will be minimized.
[INFO]: Extracting core dimensions...
[INFO]: Set CORE_WIDTH to 88.78, CORE_HEIGHT to 62.56.
```
which is much smaller than what I set.

while as I set `FP_SIZING "absolute"` and `DIE_AREA "0 0 250 300"` the step 3 shows

```
[STEP 3]
[INFO]: Running Initial Floorplanning...
[INFO]: Extracting core dimensions...
[INFO]: Set CORE_WIDTH to 238.74, CORE_HEIGHT to 277.44.
```
which is relatively close as what I set. 

I know the difference between die area and core area, i.e. die area is larger than the core area. Therefore, I want to know what is/how to set the ratio between die area and core area, did I miss something needed to be set in `config.tcl`? Additionally, how to set the height vs. width ratio? Will the large die area cause trouble on XOR step?

I will appreciate if there is anyone who can point me to a right direction. Thanks!

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v4.15.0-189-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: a9883335a102c9ba1d862a209a72fc8b4c20c785
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
--- 
Git Log (Last 3 Commits)

a988333 2022-06-18T11:53:39+02:00 Enhancements + Bugfixes to Multi-PDK Support (#1138) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.19_01.37.45)
732c856 2022-06-18T11:22:57+02:00 Fix `skip_basic_prep`, step mechanics (#1143) - Mohamed Gaber -  ()
074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (tag: 2022.06.15_01.35.48)
```

### Reproduction Material
The OpenLane did not produce `openlane_issue_repo` directory.
[repo_materials.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9270580/repo_materials.tar.gz)

### Expected behavior
The large should pass the XOR and the flow should be successful.

### Logs

This is the error message which shows the XOR was killed.

```
[STEP 33]
[INFO]: Running XOR on the layouts using Klayout...
[ERROR]: during executing: "bash /openlane/scripts/klayout/xor.sh /openlane/designs/fpga_core/runs/RUN_2022.08.05_00.19.14/results/signoff/fpga_core.gds /openlane/designs/fpga_core/runs/RUN_2022.08.05_00.19.14/results/signoff/fpga_core.klayout.gds fpga_core /openlane/designs/fpga_core/runs/RUN_2022.08.05_00.19.14/reports/signoff/fpga_core.xor.gds |& tee /dev/null /openlane/designs/fpga_core/runs/RUN_2022.08.05_00.19.14/logs/signoff/33-xor.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
--- Running XOR for 67/20 ---
"input" in: xor.drc:38
    Polygons (raw): 7145806 (flat)  16837 (hierarchical)
    Elapsed: 0.010s  Memory: 847.00M
"input" in: xor.drc:38
    Polygons (raw): 114623889 (flat)  17315 (hierarchical)
    Elapsed: 0.010s  Memory: 847.00M
"^" in: xor.drc:38
/openlane/scripts/klayout/xor.sh: line 35:  1033 Killed                  klayout -b -r $(dirname $0)/xor.drc -rd top_cell=$3 -rd a=$1 -rd b=$2 -rd thr=$(nproc) -rd ol=$4 -rd o=$4 -rd ext=${4##*.}
child process exited abnormally

[ERROR]: Flow failed.
```

vijayank88, 2022-08-06:
@AyaseErii 
Is this mpw design trying with caravel flow or testing with OpenLane flow directly?
Try to update the OpenLane to latest commit and let us know. [2022.07.30](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.07.30)

vijayank88, 2022-08-06:
@AyaseErii 
Attachment contains only logs. Share def file or verilog source files to debug the issue.

AyaseErii, 2022-08-06:
Thanks for the reply! I am testing with OpenLane flow directly. I found the reason: the shortage of the memory killed the XOR implementation. I tried to move all tools and designs to another server with larger memory space, the flow passes the XOR and works all the way to the end.

AyaseErii, 2022-08-06:
Now the issue is that: the actual core size of the design will mismatch the settings in the `config.tcl` significantly for the small design. 

Say I set the absolute die area
```
set ::env(FP_SIZING) "absolute" 
set ::env(DIE_AREA) "0 0 100 86"
```
Then it shows the warning with a significantly reduced core size:

```
[STEP 3]
[INFO]: Running Initial Floorplanning...
[WARNING]: Current core area is too small for a power grid. The power grid will be minimized.
[INFO]: Extracting core dimensions...
[INFO]: Set CORE_WIDTH to 88.78, CORE_HEIGHT to 62.56.
```

Then If I set for the same small design
```
set ::env(FP_SIZING) "absolute" 
set ::env(DIE_AREA) "0 0 250 300"
```
The tool will set the core area much close to the die area:
```
[STEP 3]
[INFO]: Running Initial Floorplanning...
[INFO]: Extracting core dimensions...
[INFO]: Set CORE_WIDTH to 238.74, CORE_HEIGHT to 277.44.
```

How can control precisely when I use a small design? Did I miss something in the `config.tcl` file? For example, I want to set the `DIE_AREA "0 0 110 92"` since I need a core area like `99 83` which is 90% of the die area, what command should I add to control these?

Thanks!

AyaseErii, 2022-08-10:
Hi All,

Thanks for the reply and answers. After doing trial and errors many times, I believe the issue was caused by inappropriate PDN pitch setups. In order to get the small design works (with a fixed die number), I need to set the `FP_PDN_VPITCH` and `FP_PDN_HPITCH` with relatively smaller value rather than the default values which I did not set at all. Now the small design works well with my settings. However, by this way, it caused some trouble/errors when I was trying to harden the small macros. I will close this issue, and open another issue regarding how to match/align the power grids set by users with that of the small macros.

Thanks!

> Now the issue is that: the actual core size of the design will mismatch the settings in the `config.tcl` significantly for the small design.
> 
> Say I set the absolute die area
> 
> ```
> set ::env(FP_SIZING) "absolute" 
> set ::env(DIE_AREA) "0 0 100 86"
> ```
> 
> Then it shows the warning with a significantly reduced core size:
> 
> ```
> [STEP 3]
> [INFO]: Running Initial Floorplanning...
> [WARNING]: Current core area is too small for a power grid. The power grid will be minimized.
> [INFO]: Extracting core dimensions...
> [INFO]: Set CORE_WIDTH to 88.78, CORE_HEIGHT to 62.56.
> ```
> 
> Then If I set for the same small design
> 
> ```
> set ::env(FP_SIZING) "absolute" 
> set ::env(DIE_AREA) "0 0 250 300"
> ```
> 
> The tool will set the core area much close to the die area:
> 
> ```
> [STEP 3]
> [INFO]: Running Initial Floorplanning...
> [INFO]: Extracting core dimensions...
> [INFO]: Set CORE_WIDTH to 238.74, CORE_HEIGHT to 277.44.
> ```
> 
> How can I control precisely when I use a small design? Did I miss something in the `config.tcl` file? For example, I want to set the `DIE_AREA "0 0 110 92"` since I need a core area like `99 83` which is 90% of the die area, what command should I add to control these?
> 
> Thanks!

closed 2022-08-10 by AyaseErii