The-OpenROAD-Project/OpenLane #1257: What is the maximum size supported by the openlane tool? Including cels, placements, routing, etc.

pengpeng-lian, 2022-08-09:
What is the maximum size supported by the openlane tool? Including cels, placements, routing, etc.

vijayank88, 2022-08-09:
@pengpeng-lian 
OpenLane currently supports 130nm node.
I don't get any meaning for  `maximum size support`

donn, 2022-08-09:
We don't have any size limits I'm aware of. @maliberty? Is there a practical maximum to the die area?

maliberty, 2022-08-09:
Its really just limited by memory and runtime.  In a flat run 1-2M instances is generally the practical max even with proprietary tools.  That assumes a big machine and a lot of patience not a laptop.

donn, 2022-08-09:
Well, there you have it, @pengpeng-lian.

As far as I personally know, the largest design ever hardened with OpenLane was 200k cells and took 9 hours.

closed 2022-08-09 by donn