The-OpenROAD-Project/OpenLane #1260: [PDN issue] How to match/align the power grids of macros with the power grids of the die

AyaseErii, 2022-08-10:
### Description
Hi, I am trying to do a manual macro placement with OpenLane. When the macros are relatively large, the manual macro placement was implemented well. However, with the same top design, I made each macro to be placed relatively smaller, i.e. changed the `FP_PDN_VPITCH` and `FP_PDN_HPITCH` to smaller value like `10` or `20`, then I met the errors shown in the end. I believe it is the issue of mismatching the power grids between small macros and the whole die. Since I try to set the pitch values to make the grids to be aligned but it still showed some errors. May I know how to solve such a mismatching issue? Any suggestions? Thanks!

### Environment
```
python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v4.15.0-189-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 4476a58407d670d251aa0be6a55e5391bb181c4e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (HEAD -> master, tag: 2022.07.30, origin/master, origin/HEAD)
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  (tag: 2022.07.28)
0b55101 2022-07-26T20:37:20+02:00 [BOT] Update openroad_app (#1219) - Openlane Bot -  (tag: 2022.07.27)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

```

### Reproduction Material
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9301301/issue_reproducible.zip)

### Expected behavior
The manual macro placement with small macro designs should pass the PDN step and be implemented without error all the way to the end.

### Logs
```
[STEP 7]
[INFO]: Generating PDN...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/fpga_core/runs/RUN_2022.08.10_02.40.12/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
[INFO ODB-0130]     Created 78 pins.
[INFO ODB-0131]     Created 10431 components and 29537 component-terminals.
[INFO ODB-0133]     Created 979 nets and 1840 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/fpga_core/runs/RUN_2022.08.10_02.40.12/results/floorplan/fpga_core.def
[INFO PDN-0001] Inserting grid: stdcell_grid
[WARNING PDN-0178] Remaining channel (5.2800, 540.0000) - (994.3000, 580.0000) on met4 for nets: VGND
[WARNING PDN-0178] Remaining channel (99.7200, 540.0000) - (821.3200, 580.0000) on met4 for nets: VGND
[ERROR PDN-0179] Unable to repair all channels.
PDN-0179
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

maliberty, 2022-08-10:
Without looking at your test case I would guess you have channels between your macros that are too small.  Would you include a screen shot around the areas reported in the warnings.

AyaseErii, 2022-08-10:
> Without looking at your test case I would guess you have channels between your macros that are too small. Would you include a screen shot around the areas reported in the warnings.

Thanks for the reply! Sure, please find the screenshots below:
```
[WARNING PDN-0178] Remaining channel (5.2800, 540.0000) - (994.3000, 580.0000) on met4 for nets: VGND
```
[image]

```
[WARNING PDN-0178] Remaining channel (99.7200, 540.0000) - (821.3200, 580.0000) on met4 for nets: VGND
```
[image]

Actually, I do not quite understand what/where the "channels" are. The screenshots are just taken based on the indication of the warning messages. Are the "channels" gaps between macros or something else?

maliberty, 2022-08-10:
The channel is the region between macros.  I've highlighted it:
[image]

I think this dimension is too small and you can't fit a power strap in there.  If you spread the macros a bit further apart in Y I think it will resolve the issue.

AyaseErii, 2022-08-17:
> The channel is the region between macros. I've highlighted it: [image]
> 
> I think this dimension is too small and you can't fit a power strap in there. If you spread the macros a bit further apart in Y I think it will resolve the issue.

Thanks! I increased the spacing but stuck on some issue in PDN step for a long time. Finally I find a way to solve the issue of PDN by using advanced power grid settings.

closed 2022-08-17 by AyaseErii

jessimd, 2022-08-17:
Hi, I am experiencing a similar error and was wondering what steps you took to solve the issue using the advanced power grid settings. Thanks!

Baungarten-CINVESTAV, 2022-11-18:
Hello, Could you help me with these questions?

- What did you do to generate these images?

- I mean, how can I see image like that of my own design?

maliberty, 2022-11-18:
@Baungarten-CINVESTAV You can open the design in the OR gui.  I don't know which flow you are using but the basic function is "openroad -gui" and then load your design.