The-OpenROAD-Project/OpenLane #1262: [PDNGEN Issue] -  PDN does not honor the FP_PDN_HORIZONTAL_HALO & FP_PDN_VERTICAL_HALO value

dineshannayya, 2022-08-11:
### Description

My openlane flow global stage fails with DRC failure, further debug shows that PDNGen does not honor the minimum spacing margin specification for Macro boundary and connecting the MET4 Power Strip near to MACRO Boundary, Due to this global stage fails with DRC failure.  FYI:  SRAM Macro has Pin at MET4.  

[image]

 
### Environment
<pre>python3 ./env.py issue-survey
Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 6ab944bc23688cae6dc6fa32444891a1e57715c8
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

6ab944b 2022-08-09T14:51:35+02:00 IR Drop Reports in the Signoff Stage (#1255) - Mohamed Gaber -  (HEAD -&gt; master)
4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  ()
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

### Reproduction Material
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9310642/issue_reproducible.tar.gz)

### Expected behavior
PDNGen need to honor the minimum spacing margin specification for Macro boundary 

### Logs
There is no failure at PDN stage, but openlane flow failure at global routing phase.  I have created test-setup at after PDN stage

arlpetergadfort, 2022-08-11:
@dineshannayya I think I see what the issue is here and have a PR opening soon.

arlpetergadfort, 2022-08-11:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2139

arlpetergadfort, 2022-08-11:
@dineshannayya this has been merged in so there should be honoring of the halo now

donn, 2022-08-12:
@arlpetergadfort OpenROAD updates are not automatically merged into OpenLane. I still need to pull it in.

arlpetergadfort, 2022-08-12:
> @arlpetergadfort OpenROAD updates are not automatically merged into OpenLane. I still need to pull it in.

@donn I don't use OpenLane, so it will be on you and the maintainers it to update OpenROAD for OpenLane

dineshannayya, 2022-08-14:
@arlpetergadfort My issue is resolved with locally building openlane docker built with your openroad  changes. Appreciate for quick follow-up.

dineshannayya, 2022-08-15:
@arlpetergadfort Even though Openlane flow is clean with latest PDN changes, I see uniformity of the Vertical power strip are missing.  This may create IR Drop issue ..

[image]

arlpetergadfort, 2022-08-15:
> @arlpetergadfort Even though Openlane flow is clean with latest PDN changes, I see uniformity of the Vertical power strip are missing. This may create IR Drop issue ..
> 
> [image]

I guess the question is: does it cause IR drop issues? Please check the impact before raising the issue.
The reason they are missing is PDNgen removed them since they were only connected to 1 or fewer straps/pins, therefore they do not add to the grid in any meaningful way.

donn, 2022-08-15:
> @donn I don't use OpenLane, so it will be on you and the maintainers it to update OpenROAD for OpenLane

Then please refrain from proclaiming that something has been "merged in" in OpenLane issues. It may confuse users.

arlpetergadfort, 2022-08-15:
@donn I will, I clearly meant that it was integrated into OR and @dineshannayya (clearly understood).

PR #1267 (2022-08-16): Update OpenROAD
~ openroad_app -> `4a99e88667b0840531ca0096c4fa0da8f80d6cb1`
~ Updated `repair_antennas` invocation: see https://github.com/The-OpenROAD-Project/OpenROAD/pull/2136#issuecomment-1212065257

---

Resolves #1264
Resolves #1262
closed 2022-08-16 by donn