The-OpenROAD-Project/OpenLane #1263: init_design_config creates directory in the wrong path

armleo, 2022-08-12:
### Description
When running:
```
./flow.tcl -design mem_1r1w -init_design_config
```
It creates `$OPENLANE_ROOT/openlane/mem_1r1w` instead of  `$OPENLANE_ROOT/designs/mem_1r1w`

### Environment
OpenLane installation from commit: 6ab944bc23688cae6dc6fa32444891a1e57715c8

### Reproduction Material
Not relevant

### Expected behavior
It creates `$OPENLANE_ROOT/openlane/mem_1r1w` instead of  `$OPENLANE_ROOT/designs/mem_1r1w`

### Logs

```
It creates `$OPENLANE_ROOT/openlane/mem_1r1w` instead of  `$OPENLANE_ROOT/designs/mem_1r1w`
```

vijayank88, 2022-08-12:
@armleo 
The latest version of flow updated to add new designs into OpenLane flow by using following command.
```
./flow.tcl -design <design_name> -init_design_config -add_to_designs
```
Can you try with above command and let us know, is still creating new `openlane` directory for adding `designs`?

armleo, 2022-08-12:
@vijayank88 

Thanks for the response. Yeah, I assumed that the behavior stayed the same. But the documentation clearly states the behavior.

closed 2022-08-12 by armleo