The-OpenROAD-Project/OpenLane #1264: Latest Openlane Global Routing stage fails with [ERROR GRT-0069] Error

dineshannayya, 2022-08-13:
### Description
With latest Openlane flow, some of the sub-macro of Riscduino blocks hardening fails with [ERROR GRT-0069] . These blocks were successfully harden with MPW-5/MPW-6 tool set. I see these block are part of Openlane Regression [https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/tree/c72552a358466ecb13309ed26ec5993f57aba473/scripts/rdS3_ycr_core_top] But look like they are harden by 16 day older Openlane.

### Environment
```
<pre>python3 ./env.py issue-survey
Kernel: Linux v5.15.0-41-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 6ab944bc23688cae6dc6fa32444891a1e57715c8
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

6ab944b 2022-08-09T14:51:35+02:00 IR Drop Reports in the Signoff Stage (#1255) - Mohamed Gaber -  (HEAD -&gt; master)
4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  ()
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>
```

### Reproduction Material
[ycr_core_top_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9330718/ycr_core_top_issue_reproducible.tar.gz)

### Expected behavior
Looking for successful hardening of the block. 

### Logs
```
<pre>[STEP 17]
[INFO]: Running Global Routing...
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Log: ycr_core_top/runs/ycr_core_top/logs/routing/17-global.log
[ERROR]: Last 10 lines:
met3             86034          4780            5.56%             0 /  0 /  0
met4             53727         18847           35.08%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total           416380        167870           40.32%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 1586482 um
[INFO GRT-0014] Routed nets: 22474
[ERROR GRT-0069] Diode cell sky130_fd_sc_hd__diode_2/DIODE not found.
Error: groute.tcl, 60 GRT-0069
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager
</pre>
```

vijayank88, 2022-08-13:
@dineshannayya 
With latest OpenROAD commit i am not facing any issue. GRT completed successfully.
```
 report_design_area
============================================================================
Design area 228429 u^2 46% utilization.
area_report_end
openroad> 
```
OpenROAD version I used: `v2.0-4602-g3048eee5f`

vijayank88, 2022-08-13:
@dineshannayya 
`[ERROR GRT-0069] Diode cell sky130_fd_sc_hd__diode_2/DIODE not found.` is that causing issue at your end any issue?

dineshannayya, 2022-08-13:
@vijayank88 All the diodes are inserted by the openlane flow

dineshannayya, 2022-08-13:
I am using latest openroad version:  OpenROAD b57893514040c5aa453ab5c104f35d1173a8add9 to take care latest PDN Bug fix https://github.com/The-OpenROAD-Project/OpenLane/issues/1262,  I see there is some grt fix done in latest version

I am using latest below Openroad version:
---------------------------------------------------------------------------------
commit b57893514040c5aa453ab5c104f35d1173a8add9
Merge: 8de639048 3a1d5329c
Author: Eder Monteiro <43450810+eder-matheus@users.noreply.github.com>
**Date:   Thu Aug 11** 19:50:48 2022 -0300

    **Merge pull request #2143 from eder-matheus/grt_fix_instability**
--------------------------------------------------------------------------------------------

You are using 

commit 3048eee5f7448f8048e803e6f776a9331bfe7913
Merge: 8ea622456 31b64d603
Author: Osama Hammad <56893454+osamahammad21@users.noreply.github.com>
**Date:   Wed Aug 10** 23:59:40 2022 +0200

    Merge pull request #2120 from osamahammad21/debug_single_worker_fix

vijayank88, 2022-08-13:
Ok. Let me merge latest commit from OpenROAD and try once again and update on this.

vijayank88, 2022-08-13:
@dineshannayya 
yes this issue confirmed with OpenROAD commit: https://github.com/The-OpenROAD-Project/OpenROAD/commit/988b581071baca1852e654fe8fa2cbf22296f804

maliberty, 2022-08-13:
@vijayank88 note that in the head of OR it is no longer necessary to give the diode pin, just the cell name

dineshannayya, 2022-08-14:
@vijayank88 With above groute.tcl change my flow is clean-now.  Thanks

PR #1267 (2022-08-16): Update OpenROAD
~ openroad_app -> `4a99e88667b0840531ca0096c4fa0da8f80d6cb1`
~ Updated `repair_antennas` invocation: see https://github.com/The-OpenROAD-Project/OpenROAD/pull/2136#issuecomment-1212065257

---

Resolves #1264
Resolves #1262
closed 2022-08-16 by donn