The-OpenROAD-Project/OpenLane #1265: OpenROAD GUI corrupts it's own memory

armleo, 2022-08-13:
### Description
I am trying to open OpenROAD GUI. When I open some stuff, after some time I get very weird behaviour:

[image]

As you can see the menu is corrupt. In fact, there is more issues than that. Closing the OpenROAD gui gives an error mesage with corrupt content. There is a lot of unhandled events causing many errors in the log. Cant reproduce them, as behavour changes frequently.

It's not clear what causes this. The host is Manjaro, but the issue is reproducible in Ubuntu across many machines. The OpenROAD gui is from Docker image.

### Environment
Commit: 6ab944b
Everything is fine. Not supplying the env info, since it contains NDA info.

### Reproduction Material
Not relevant

### Expected behavior
OpenROAD gui not corrupting it's own GUI

### Logs
Not provided.

QuantamHD, 2022-08-14:
So there's a few things in there that make me feel like this is not an OpenROAD specific issue. If you look at the artifacts they overlap the gui elements. 

The canvas is rendered to its own framebuffer which would prevent it from modifying elements outside the canvas element. This looks like either a driver issue, or faulty cable.

I could be wrong, but the first thing I'd try is to get a new video cable, and see if it stops the issue. If not try updating your video drivers.

armleo, 2022-08-14:
> So there's a few things in there that make me feel like this is not an OpenROAD specific issue. If you look at the artifacts they overlap the gui elements.

Its the OpenROAD GUI that is open, that's why it is kind of overlapping.

> The canvas is rendered to its own framebuffer which would prevent it from modifying elements outside the canvas element. This looks like either a driver issue, or faulty cable.

Faulty cable is not the cause. I would not be able to capture it using screenshot. Driver issue might be the case, but it's running inside Docker. If there is a driver issue then I need help debugging it and pinpointing it, I guess?

armleo, 2022-08-14:
Able to reproduce it in another branch with no NDA info (ignore the commits, the modify only documentation files). It is based on commit: https://github.com/The-OpenROAD-Project/OpenLane/commit/6ab944bc23688cae6dc6fa32444891a1e57715c8

```
python3 ./env.py issue-survey
Kernel: Linux v5.9.16-1-MANJARO
Distribution: manjaro 21.3.0
Python: v3.10.5 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: merge-window-4-dev
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c432d9b 2022-08-14T01:11:08+04:00 Digital Guide WIP - Arman Avetisyan -  (HEAD -> merge-window-4, origin/merge-window-4)
75d61ff 2022-08-12T08:33:16+04:00 Moved digital flow to digital intro - Arman Avetisyan -  ()
b95499e 2022-08-12T08:30:49+04:00 Reworked structure of docs - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/armleo/OpenLane.git (fetch)
origin	https://github.com/armleo/OpenLane.git (push)
upstream	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
upstream	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

maliberty, 2022-08-14:
What data are you reproducing it on?  A commit is not enough to take any action.

When you say you reproduce it on Ubuntu, are you running the X server on Ubuntu or is it still displaying on Manjaro?  What version of Ubuntu?

armleo, 2022-08-14:
> What data are you reproducing it on? A commit is not enough to take any action.

I will post the data in a second.

> When you say you reproduce it on Ubuntu, are you running the X server on Ubuntu or is it still displaying on Manjaro? What version of Ubuntu?

Ubuntu inside VirtualBox with installed 3D Acceleration extension. Able to reproduce it across many (VM) machines from versions 20.04 up to 21.10.

armleo, 2022-08-14:
Environment:

[image]

Steps:
```
# Running inside VirtualBox with 3D Acceleration driver from VirtualBox extensions
# 

git clone --depth 1 https://github.com/The-OpenROAD-Project/OpenLane.git OpenLane_OpenROAD_gui_issue

cd OpenLane_OpenROAD_gui_issue

make; make test

make; make test

openroad -gui

# inside the gui:

read_lef $::env(PDK_ROOT)/sky130B/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef
read_lef designs/spm/runs/openlane_test/tmp/merged.nom.lef

read_def designs/spm/runs/openlane_test/results/final/def
```

Log:
```
make mount

cd /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue && \

	docker run --rm -v /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue:/openlane -v /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/designs:/openlane/install -v /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/pdks:/home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/pdks -e PDK_ROOT=/home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/pdks -e PDK=sky130A -e STD_CELL_LIBRARY=sky130_fd_sc_hd  --user 1000:1000 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/armleo/.Xauthority:/.Xauthority --network host -ti efabless/openlane:6ab944bc23688cae6dc6fa32444891a1e57715c8-amd64

OpenLane Container (6ab944b):/openlane$ openroad -gui

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'

libGL error: unable to load driver: swrast_dri.so

libGL error: failed to load driver: swrast

OpenROAD e036ecfaca4bc0efe88a54085efcf0f562c48a6b 

This program is licensed under the BSD-3 license. See the LICENSE file for details.

Components of this program may be licensed under more restrictive licenses which must be honored.

[INFO ODB-0222] Reading LEF file: /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/pdks/sky130B/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef

[INFO ODB-0223]     Created 13 technology layers

[INFO ODB-0224]     Created 25 technology vias

[INFO ODB-0226] Finished LEF file:  /home/armleo/Desktop/OpenLane_OpenROAD_gui_issue/pdks/sky130B/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef

[INFO ODB-0222] Reading LEF file: designs/spm/runs/openlane_test/tmp/merged.nom.lef

[WARNING ODB-0217] duplicate VIARULE (L1M1_PR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (L1M1_PR_R) ignoring...

[WARNING ODB-0217] duplicate VIARULE (L1M1_PR_M) ignoring...

[WARNING ODB-0217] duplicate VIARULE (L1M1_PR_MR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (L1M1_PR_C) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M1M2_PR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M1M2_PR_R) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M1M2_PR_M) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M1M2_PR_MR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M1M2_PR_C) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M2M3_PR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M2M3_PR_R) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M2M3_PR_M) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M2M3_PR_MR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M2M3_PR_C) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M3M4_PR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M3M4_PR_R) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M3M4_PR_M) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M3M4_PR_MR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M3M4_PR_C) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M4M5_PR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M4M5_PR_R) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M4M5_PR_M) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M4M5_PR_MR) ignoring...

[WARNING ODB-0217] duplicate VIARULE (M4M5_PR_C) ignoring...

[WARNING ODB-0220] WARNING (LEFPARS-2036): SOURCE statement is obsolete in version 5.6 and later.

The LEF parser will ignore this statement.

To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file designs/spm/runs/openlane_test/tmp/merged.nom.lef at line 930.

[INFO ODB-0225]     Created 441 library cells

[INFO ODB-0226] Finished LEF file:  designs/spm/runs/openlane_test/tmp/merged.nom.lef

[INFO ODB-0127] Reading DEF file: designs/spm/runs/openlane_test/results/final/def/spm.def

[INFO ODB-0128] Design: spm

[INFO ODB-0130]     Created 38 pins.

[INFO ODB-0131]     Created 1219 components and 5708 component-terminals.

[INFO ODB-0132]     Created 2 special nets and 4666 connections.

[INFO ODB-0133]     Created 380 nets and 1042 connections.

[INFO ODB-0134] Finished DEF file: designs/spm/runs/openlane_test/results/final/def/spm.def

[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.

[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.

[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.

```

Result:
[image]

Data:
[runs.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9332802/runs.tar.gz)

maliberty, 2022-08-14:
I just tried it with ubuntu 22.04 using wsl and have no problem:
[image]

I've used virtual box in the past without trouble either.  I expect your issue is either you accelerator or your host OS.  I don't see anything in OR to fix.

donn, 2022-08-15:
Yeah, agreed with @maliberty.

Something not working properly in VirtualBox is basically a guarantee. I would suggest people use the following setups for running GUI apps inside the Docker image:

* WSL + Docker + Xming (Windows 10)
* WSL + Docker (Windows 11)
* Docker or Lima + XQuartz (macOS)
* Docker or Podman (Linux)

closed 2022-08-15 by donn