The-OpenROAD-Project/OpenLane #1266: Relationship between FP_PDN_VERTICAL_HALO and Macro

pengpeng-lian, 2022-08-15:
I am running a project like this, which has both macro and verilog project, I set the parameters: VERILOG_FILES_BLACKBOX, EXTRA_LEFS, EXTRA_GDS_FILES, I also set the FP_PDN_HORIZONTAL_HALO, FP_PDN_VERTICAL_HALO, but check the DEF file, I will find that the DEF will be directly connected to macro, I guess he should not recognize macro, how to solve this problem？

maliberty, 2022-08-15:
I don't understand your description.  Please attach a test case and explain what you see versus what you hope to see.

pengpeng-lian, 2022-08-16:
My top level contains both macro and rtl code, I set ::env(FP_PDN_HORIZONTAL_HALO) "10"
set ::env(FP_PDN_VERTICAL_HALO) "10". But the tool does not give me PDN halo, the situation is as follows.
[image]

arlpetergadfort, 2022-08-16:
I think this is the same issue as https://github.com/The-OpenROAD-Project/OpenLane/issues/1262

pengpeng-lian, 2022-08-17:
> @pengpeng-lian would you try with the latest OR and see if it resolves your issue?

Thank you very much for your answer. I had the same problem as his and I followed this link.

closed 2022-08-17 by maliberty