The-OpenROAD-Project/OpenLane #1268: Routing Congestion Problem

Ali-Sabir2, 2022-08-16:
[image]
Hi!
**I am facing routing congestion issue,**
I did change density and area multiple time but didn't get positive response.
Thanks

vijayank88, 2022-08-16:
@Ali-Sabir2 
Missing issue template and Please package a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md and attach here

vijayank88, 2022-08-16:
From the screenshot its looks like out of memory kill. Please increase your swap space and run the flow again.

donn, 2022-08-16:
Please use the issue template. Help us help you.

closed 2022-08-16 by donn