The-OpenROAD-Project/OpenLane #1269: Magic reports DRC errors

vijayank88, 2022-08-17:
### Description
I just brief on this issue track from where its started and now struck at Magic DRC error.

This is a MPW Shuttle project https://github.com/merledu/Ghazi_DFT and 5 more design have similar issue.

This design can be found here as MPW CI regression if need more details on previous stages of the flow: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Funstable-small-1/detail/PR-72/1/pipeline/314 

1. Initially this design got failed during detail routing as contains special nets and resolved using `insert_tiecells` from OpenROAD.
2. Later its got failed during LVS as default `config.tcl` disabled to power tie cells and resolved by removing configuration from 76-86 https://github.com/merledu/Ghazi_DFT/blob/96f998f3331a1258d064294751be33765bfbd3e9/openlane/user_project_wrapper/config.tcl#L76 and passed LVS stage. Slack channel discussion here: https://open-source-silicon.slack.com/archives/C032Y8J3KHA/p1658908120910749 
3. Now its got struck for Magic DRC error:

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey

python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 7b15116b80bd715b5828d278b5be9f6ec91ebdef
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

7b15116 2022-08-15T18:27:17+02:00 Revamp Net Unbuffering (#1261) - Mohamed Gaber -  (HEAD -> master, tag: 2022.08.16, origin/master, origin/HEAD)
6ab944b 2022-08-09T14:51:35+02:00 IR Drop Reports in the Signoff Stage (#1255) - Mohamed Gaber -  (tag: 2022.08.10)
4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (tag: 2022.07.30)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

```

### Reproduction Material
* Upload a tarball containing the relevant design. 
* List the commands used to run the design.

- Unzip drc_check.zip
- move this directory inside OpenLane/designs/
- make mount
- source run.tcl
This will report DRC errors

### Expected behavior
Clean DRC report

### Logs
```
DRC style is now "drc(full)"
Loading DRC CIF style.
[INFO]: COUNT: 22652518
[INFO]: Should be divided by 3 or 4
[INFO]: DRC Checking DONE (/openlane/designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/reports/signoff/drc.rpt)
[INFO]: Saving mag view with DRC errors (/openlane/designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/results/signoff/user_project_wrapper.drc.mag)
[INFO]: Saved
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[ERROR]: during executing: "openroad -python /openlane/scripts/drc_rosetta.py tr to_klayout -o /openlane/designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/reports/signoff/drc.klayout.xml --design-name user_project_wrapper /openlane/designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/reports/signoff/drc.tr"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
OpenROAD e036ecfaca4bc0efe88a54085efcf0f562c48a6b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
child killed: kill signal

[INFO]: Calculating runtime...
[INFO]: Saving current set of views in 'designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/user_project_wrapper/runs/RUN_2022.08.16_11.04.09/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -python $::env(SCRIPTS_DIR)/drc_rosetta.py tr to_klayout -o $::env(drc_prefix).klayout.xml  --design-name $::env(DESIGN..."
    (procedure "run_magic_drc" line 19)
    invoked from within
"run_magic_drc"
    (procedure "run_drc_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)

```

donn, 2022-08-17:
Can you run `export CREATE_REPRODUCIBLE_FROM_SCRIPT=magic/drc.tcl` and give us the resulting reproducible, please?

vijayank88, 2022-08-18:
Reproducible generated through script contains file size of `378MB`, i will upload and share drive link. Thanks

donn, 2024-09-12:
Please open a new issue if this persists, thanks.

closed 2024-09-12 by donn