The-OpenROAD-Project/OpenLane #1270: Detail route issue: No access point for inst_eFPGA_top/io_in[10]

vijayank88, 2022-08-17:
### Description
@maliberty 
This is MPW shuttle design. https://github.com/nguyendao-uom/eFPGA_v3_caravel

Seems the macro covered entire die area and Triton route unable to access those pins. 

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 7b15116b80bd715b5828d278b5be9f6ec91ebdef
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

7b15116 2022-08-15T18:27:17+02:00 Revamp Net Unbuffering (#1261) - Mohamed Gaber -  (HEAD -> master, tag: 2022.08.16, origin/master, origin/HEAD)
6ab944b 2022-08-09T14:51:35+02:00 IR Drop Reports in the Signoff Stage (#1255) - Mohamed Gaber -  (tag: 2022.08.10)
4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (tag: 2022.07.30)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction Material
* Upload a tarball containing the relevant design. 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9359201/issue_reproducible.zip)

* List the commands used to run the design.

### Expected behavior
Whether Tritonroute able to complete the routing or mark this as floorplan issue from designer?

### Logs
```
[STEP 23]
[INFO]: Running Detailed Routing (log: logs/routing/23-detailed.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Log: designs/eFPGA_v3_wrapper/runs/RUN_2022.08.17_11.30.46/logs/routing/23-detailed.log
[ERROR]: Last 10 lines:
[INFO DRT-0033] met3 shape region query size = 105912.
[INFO DRT-0033] via3 shape region query size = 19506.
[INFO DRT-0033] met4 shape region query size = 15034.
[INFO DRT-0033] via4 shape region query size = 2448.
[INFO DRT-0033] met5 shape region query size = 1029.
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 100 pins.
[ERROR DRT-0073] No access point for inst_eFPGA_top/io_in[10].
Error: droute.tcl, 46 DRT-0073
child process exited abnormally

```

maliberty, 2022-08-17:
With the head (b983eb1e521644fd33877c14ccd5cdde2b99aa89) I don't get any PA issue.

This design looks nothing like an FPGA:
[image]

are you sure you attached the right test case?

donn, 2022-08-17:
OpenROAD has been updated since the OpenLane version you're currently using- please update to the latest version of OpenLane and try again.

vijayank88, 2022-08-18:
@maliberty 
I've tested `issue_reproducible` locally before uploading.
The design i shared contains only a macro and top level connection, you can see screenshot in description.
Attached the test case again.

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9367160/issue_reproducible.zip)

maliberty, 2022-08-18:
This is a bad LEF for this macro and the pin is inaccessible.  In the LEF you have:
```
LAYER met4
  SPACINGTABLE
     PARALLELRUNLENGTH 0
     WIDTH 0 0.3
     WIDTH 3 0.4 ;
```
So 0.4 applies to anything with a width > 3.  The creator chose to do a pin cutout:
[image]

so the blockage triggers the 0.4 rule but the cutout is only 0.3.  It is unnecessary to do a cutout at all in the case of a pin on the edge of the macro like this as any covering blockage is automatically handled.

closed 2022-08-18 by maliberty

vijayank88, 2022-08-18:
@maliberty 
So user has to update the LEF file by re-harden the macro with latest tool or I can change LEF file value for the pins width to complete routing stage?

maliberty, 2022-08-18:
You can update the LEF if you wish.  I'm not clear why this would ever have worked but perhaps there was some bug that allowed it.

vijayank88, 2022-08-18:
Thanks @maliberty