The-OpenROAD-Project/OpenLane #1271: Old OpenLane version can pass the detail routing but latest version cannot

AyaseErii, 2022-08-17:
### Description
Hello, with the same design, I happened to find that the old openlane version can pass the detail routing which will make the violations to 0, while the latest openlane cannot pass and remain several violations. I checked the openroad-app commit as well. The old openlane's openroad-app commit is `0b8b7ae255f8fbbbefa57d443949b84e73eed757` while the latest openlane's openroad commit is `e036ecfaca4bc0efe88a54085efcf0f562c48a6b`. Therefore, I suspect that the issue should be caused by the openroad-app, and believe the mpw-7a commit can work while the other commit cannot. If it is true, how can I find the mpw-7a based openlane and/or switch to mpw-7a with latest openlane? Simply changing the commit in openroad-app with the mpw-7a commit in the latest openlane cannot work and it will remind me to change the environment. I do not have any idea about how to change the environment. Could anyone can point me in the right direction? Thanks!

### Environment
The old environment
```
python3 ./env.py issue-survey
Kernel: Linux v4.15.0-191-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: a9883335a102c9ba1d862a209a72fc8b4c20c785
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a988333 2022-06-18T11:53:39+02:00 Enhancements + Bugfixes to Multi-PDK Support (#1138) - Mohamed Gaber -  (HEAD -> master, tag: 2022.06.19_01.37.45)
732c856 2022-06-18T11:22:57+02:00 Fix `skip_basic_prep`, step mechanics (#1143) - Mohamed Gaber -  ()
074a92b 2022-06-14T15:01:11+02:00 Fix Magic Antenna Check (#1142) - Mohamed Gaber -  (tag: 2022.06.15_01.35.48)
```

The latest environment
```
python3 ./env.py issue-survey
Kernel: Linux v4.15.0-191-generic
Distribution: ubuntu 18.04
Python: v3.7.13 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 4476a58407d670d251aa0be6a55e5391bb181c4e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  (HEAD -> master, tag: 2022.07.30, origin/master, origin/HEAD)
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  (tag: 2022.07.28)
0b55101 2022-07-26T20:37:20+02:00 [BOT] Update openroad_app (#1219) - Openlane Bot -  (tag: 2022.07.27)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction Material
No `issue_repoduction` directory was produced. Will upload if needed.

### Expected behavior
The design should pass detailed routing without violation

### Logs
With latest version
```
[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total Number of violations is 18
[ERROR]: Flow failed.
```

maliberty, 2022-08-17:
@donn is there a doc on how to use the latest OR with a given OL commit?

donn, 2022-08-17:
@maliberty Update the OR commit in `dependencies/tool_metadata.yml` and invoke `BUILD_IF_CANT_PULL=1 make openlane` in the OpenLane root folder.

---

@AyaseErii If it's not too much of a hassle, can you run `export CREATE_REPRODUCIBLE_FROM_SCRIPT=droute.tcl` before running the flow? This should force a reproducible to be generated.

AyaseErii, 2022-08-18:
> @maliberty Update the OR commit in `dependencies/tool_metadata.yml` and invoke `BUILD_IF_CANT_PULL=1 make openlane` in the OpenLane root folder.
>
> @AyaseErii If it's not too much of a hassle, can you [...]

Thanks! I reproduced the issue by using the command you provided. Plz find the attached repo file:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9367054/issue_reproducible.zip)

When I tried to invoke `BUILD_IF_CANT_PULL=1 make openlane`, it showed an error that there is no module named `click`, then I use `pip3 install click` to install but it showed an message: 
```
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: click in ./.local/lib/python3.7/site-packages (8.2.0.dev0)
Requirement already satisfied: importlib-metadata in ./.local/lib/python3.7/site-packages (from click) (4.12.0)
Requirement already satisfied: typing-extensions>=3.6.4 in ./.local/lib/python3.7/site-packages (from importlib-metadata->click) (4.3.0)
Requirement already satisfied: zipp>=0.5 in ./.local/lib/python3.7/site-packages (from importlib-metadata->click) (3.8.1)
```
I really don't know what is the issue. Any suggestions?

Thanks!

vijayank88, 2022-08-18:
@AyaseErii 
Share your mpw repo link

AyaseErii, 2022-08-18:
> @AyaseErii Share your mpw repo link

I am not sure what is the mpw repo link but I put the github repo link [here](https://github.com/AyaseErii/my_openlane_project).

maliberty, 2022-08-18:
I just tried your reproducer with the current head of OR (b983eb1e521644fd33877c14ccd5cdde2b99aa89) and it routed just fine.

vijayank88, 2022-08-18:
@AyaseErii 
You can update `dependencies/tool_metadata.yml` for `openroad_app` commit with latest one and re-build openlane and try it.
Are you going to submit this to MPW shuttle or just exploring the design with OpenLane flow?

donn, 2022-08-18:
@vijayank88  Let's not confuse users. It is not their responsibility to update OpenROAD in OpenLane, that exists as a developer mechanism.

Also, that particular step won't work anyway. 

---
@maliberty It routes just fine, it just ends up with DRC errors. Does routing "just fine" include no DRC errors?

vijayank88, 2022-08-18:
@donn 
I just checked user repo for source `config.tcl`.
User trying to use mpw-7a tag, but die size more than mpw pre-defined value.
If they're not submitting to mpw shuttle, thought suggesting to use latest OpenLane tag from master repo for quick fix.

vijayank88, 2022-08-18:
@donn 
routed `just fine` means routing completed with no DRC erros.
With latest OR, routing completed without any violations for attached `issue_reproducible`
```
[INFO DRT-0199]   Number of violations = 0.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:00, memory = 5806.96 (MB), peak = 6342.12 (MB)
```

donn, 2022-08-18:
Then yeah, I need to update OR *again*

AyaseErii, 2022-08-18:
Thanks for the help. I tried to re-build the OpenLane. However, I meet this error when I use `make` in the OpenLane directory after `git-clone` the tool.

```
make[1]: Entering directory '/home/htf6ry/OpenLane'
57f3f21d6e1d0403a88d7a7973f7447597aae160: Pulling from efabless/openlane
Digest: sha256:0b9a27fe51d8d709e77f653c39dd1244738885d0d39fdefd81a272824a33b46f
Status: Image is up to date for efabless/openlane:57f3f21d6e1d0403a88d7a7973f7447597aae160
docker.io/efabless/openlane:57f3f21d6e1d0403a88d7a7973f7447597aae160
make[1]: Leaving directory '/home/htf6ry/OpenLane'
rm -rf ./venv
python3 -m venv ./venv
Error: Command '['/home/htf6ry/OpenLane/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.
Makefile:130: recipe for target 'venv/created' failed
make: *** [venv/created] Error 1
```

I do not have any idea what happened....

AyaseErii, 2022-08-18:
The pip issue has been solved since I happened to find changing `include-system-site-packages = false` to `include-system-site-packages = true` in `OpenLane/venv/pyvenv.cfg` before using `make` will solve the issue. 

Now the issue is I cannot even pass the PDN step with the same design and same parameters (which you guys ran the flow and confirmed it can pass DRC). I put the issue_repo and error message here:

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9375647/issue_reproducible.zip)

```
[STEP 7]
[INFO]: Generating PDN (log: designs/fpga_core/runs/RUN_2022.08.18_16.14.21/logs/floorplan/7-pdn.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/fpga_core/runs/RUN_2022.08.18_16.14.21/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (190.400um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (204.000um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (217.600um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (176.800um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (190.400um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (204.000um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (217.600um, 73.440um), layer: 2.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 51 PSM-0069
child process exited abnormally
```

Any suggestions? Thanks!

maliberty, 2022-08-18:
There is a known bug in check_connectivity which is currently being worked on.  It is possible that this is a false positive though I haven't checked specifically.  Do you have any macros in your design?

maliberty, 2022-08-18:
@VidyaChhabria this may be the same as what you are looking at.

AyaseErii, 2022-08-18:
@maliberty Thanks for the reply! Yes, I believe the design includes many macros. You may find the Github design repo [here](https://github.com/AyaseErii/my_openlane_project).

maliberty, 2022-08-18:
Then I suspect this should wait for @VidyaChhabria fix.  Until then I suggest you just turn off check_connectivity and rely on LVS to check.

VidyaChhabria, 2022-08-18:
This should be resolved once this [PR-2169](https://github.com/The-OpenROAD-Project/OpenROAD/pull/2169) goes through and openroad is bumped up.

VidyaChhabria, 2022-08-22:
Done.

> This should be resolved once this [PR-2169](https://github.com/The-OpenROAD-Project/OpenROAD/pull/2169) goes through and openroad is bumped up.

AyaseErii, 2022-08-22:
> Done.
> 
> > This should be resolved once this [PR-2169](https://github.com/The-OpenROAD-Project/OpenROAD/pull/2169) goes through and openroad is bumped up.

Thanks for the update! Should I update the OpenLane with the commands below?
```
    cd OpenLane/
    git checkout master
    git pull
    make 
    make test # This is to test that the flow and the pdk were properly updated
```

AyaseErii, 2022-08-22:
Hi, I did a quick test after using the update commands, but it still shows the connectivity errors.... I attached the issue_repo and the error log below:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9397526/issue_reproducible.zip)

```
[STEP 7]
[INFO]: Generating PDN (log: designs/fpga_core/runs/RUN_2022.08.22_20.38.54/logs/floorplan/7-pdn.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/fpga_core/runs/RUN_2022.08.22_20.38.54/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (190.400um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (204.000um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (217.600um, 78.880um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (176.800um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (190.400um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (204.000um, 73.440um), layer: 2.
[WARNING PSM-0038] Unconnected PDN node on net VPWR at location (217.600um, 73.440um), layer: 2.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 51 PSM-0069
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

maliberty, 2022-08-22:
A fix for the connectivity checker issue just went in today to OR.

donn, 2022-08-31:
Should be okay now- feel free to re-open

PR #1296 (2022-08-31): Update OpenROAD, Magic + DEF DRC Enhancement
~ Added `-noblockage` to magic DEF read during DRC
~ Fixed automatic tool update CI
~ magic -> 5d51e10fb969b31e6e95b5fb78d21efeccc73c14
~ openroad_app -> 4174c3ad802d2ac1d04d387d2c4b883903f6647e

---
Partial resolution to #695
Resolves #1271
closed 2022-08-31 by donn