The-OpenROAD-Project/OpenLane #1272: issue_reproducible generated with Magic DRC stage logs mismatch

vijayank88, 2022-08-18:
### Description
`issue_reproducible` generated during magic drc check returns mismatch logs/reports between standard flow and reproducible logs.

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 12a7e9ccd497950b07bf5aa1c19e34655148d60a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

12a7e9c 2022-08-17T14:09:28+05:30 [BOT] Update OpenLane tag. (#72) - github-actions[bot] -  (HEAD -> main, origin/main, origin/HEAD)
3feaa36 2022-08-15T10:17:07-03:00 Unstable s01 (#71) - vijayan -  ()
42d70a1 2022-08-11T12:49:59+05:30 [BOT] Update OpenLane tag. (#70) - github-actions[bot] -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (push)

```

### Reproduction Material
* Upload a tarball containing the relevant design.
* Due to file size limit uploaded here: https://slack-files.com/T01699QAZBQ-F03TYEQENF8-e1eda559ee
* List the commands used to run the design.

### Expected behavior
`issue_producible` should generate same drc report

### Logs
Standard flow logs during magic drc check
```
[STEP 34]
[INFO]: Running Magic DRC (log: logs/signoff/34-drc.log)...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 1592
[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

```

With `issue_reproducible`
```
OpenLane Container (c0be543):/openlane/designs/vsdbabysoc_wrapper/runs/RUN_2022.08.18_07.25.42/issue_reproducible$ ./run.sh 

Magic 8.3 revision 315 - Compiled on Fri Jul 22 13:24:23 UTC 2022.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Using technology "minimum", version 0.0
Don't know how to read GDS-II:
Nothing in "cifinput" section of tech file.
[INFO]: Loading user_project_wrapper

File user_project_wrapper.mag couldn't be read
No such file or directory
Creating new cell
"drc(full)" is not one of the DRC styles Magic knows.
The current style is "default".
The DRC styles are: default.
No errors found.
[INFO]: COUNT: 0
[INFO]: Should be divided by 3 or 4
[INFO]: DRC Checking DONE (./reports/signoff/drc.rpt)
[INFO]: Saving mag view with DRC errors (./results/signoff/user_project_wrapper.drc.mag)
[INFO]: Saved

```

vijayank88, 2022-08-18:
Also this is another Magic DRC check failed
With OpenLane tag [2022.08.16](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.08.16) its passed DRC stage.
With OpenLane tag [2022.08.17](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.08.17) flow failed with Magic DRC errors.

vijayank88, 2022-08-18:
respective CI logs can be found here:
2022.08.16 tag: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/PR-72/1/pipeline/443
2022.08.17 tag: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/PR-74/1/pipeline/497

vijayank88, 2022-08-18:
@maliberty 
For both the OpenLane tags, i see changes wrt to `OpenROAD` commit only. 
```
user_project_wrapper
----------------------------------------
Metal4 > 3um spacing to unrelated m4 < 0.4um (met4.5b)
----------------------------------------
 1024.640um 1885.520um 1024.970um 1887.360um
 1024.640um 1885.200um 1024.680um 1885.520um
Metal4 spacing < 0.3um (met4.2)
 1024.340um 2837.200um 1024.570um 2837.520um
 1024.340um 2832.080um 1024.570um 2837.200um
 1024.340um 2831.760um 1024.570um 2832.080um
 1024.340um 2831.040um 1024.570um 2831.760um
----------------------------------------
[INFO]: COUNT: 1592
[INFO]: Should be divided by 3 or 4
```
Where this this issue to be tracked now?

maliberty, 2022-08-18:
@vijayank88 this problem seems to be about building a proper reproducer.  Is there an issue for OR here?

vijayank88, 2022-08-19:
@maliberty 
Till detail routing (OpenROAD) there is no issue. 
Only during signoff stage, Magic througs drc error 1592 with `Metal4` spacing.

vijayank88, 2022-08-19:
Here's magic DRC error scrrenshot:
[image]

[image]

donn, 2022-08-22:
Let's not conflate issues, ok? The reproducer is one issue and the actual DRC violation is another issue. Trying to tackle both in the same ticket creates the kind of confusion I see here.

WRT to the reproducible, that's for OpenLane, i.e., myself to fix.

Please create a new issue for the DRC issues.

vijayank88, 2022-08-23:
Created new issue for Magic DRC issue #1282

closed 2022-09-06 by donn

tdtu1002, 2023-11-27:
have you fixed it ? and can you help me with similar errors ? i have the announcement like you?