The-OpenROAD-Project/OpenLane #1274: or_gui should load as much data as available

armleo, 2022-08-18:
### Prompt
* or_gui can be used to debug a lot of issues. It may need a little bit of a revamp to support liberty files loading, spef loading, etc.
* Make or_gui load data depending on the stage of the run.
* Do not overwrite the files (LEFs for example) by calling the prep command. Or modify prep so it only creates the file if it is not the gui.
* It is also suffering since people do not even know about its existence. Maybe mention it in the documentation?

maliberty, 2022-08-18:
I agree.  In ORFS we have a set of gui targets that loads the gui with all necessary files from any flow stage.

armleo, 2022-08-28:
Flow does not generate the SDF files in sta multi corner. There is no current_sdf that can be loaded from single point. There is no one common "libs" variable, so i have to load .LIB depending on the current stage. Overall there seems there is more work than I expected, therefore I am pausing working on this.

kareefardi, 2023-04-18:
This is handled now by multiple PRs. I believe mainly https://github.com/The-OpenROAD-Project/OpenLane/pull/1244 and https://github.com/The-OpenROAD-Project/OpenLane/pulls?q=is%3Apr+gui

closed 2023-04-18 by kareefardi