The-OpenROAD-Project/OpenLane #1280: Local Install Script not working correctly?

monibahmed, 2022-08-21:
**Command: ** python3 ./env.py local-install

Trying to install on WSL2 Ubuntu machine locally without Docker container. 

**Error:** AttributeError: 'Tool' object has no attribute 'dependencies'

**Full output:**

monib@monib-dev:~/OpenLane$ python3 ./env.py local-install
OpenLane Local Installer

    Copyright 2021-2022 Efabless Corporation. Available under the Apache License,
    Version 2.0.

    Ctrl+C at any time to quit.

    Make sure you read the documentation in ./docs/source/local_installs.md.

Which UNIX/Unix-like OS are you using? [UBUNTU-20.04/other/centos-7/arch/macos] > UBUNTU-20.04
Do you want to install dependencies using your package manager? [NO/yes] > yes
Hit:1 https://developer.download.nvidia.com/compute/cuda/repos/wsl-ubuntu/x86_64  InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal InRelease
Get:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:4 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:5 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Fetched 336 kB in 1s (371 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version (7.68.0-1ubuntu2.12).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libcairo2-dev' instead of 'libcairo-dev'
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
Note, selecting 'libncurses-dev' instead of 'ncurses-dev'
autoconf is already the newest version (2.69-11.1).
autopoint is already the newest version (0.19.8.1-10build1).
bison is already the newest version (2:3.5.1+dfsg-1).
bzip2 is already the newest version (1.0.8-2).
flex is already the newest version (2.6.4-6.2).
gawk is already the newest version (1:5.0.1+dfsg-1).
gcc is already the newest version (4:9.3.0-1ubuntu2).
gettext is already the newest version (0.19.8.1-10build1).
libcairo2-dev is already the newest version (1.16.0-4ubuntu1).
libffi-dev is already the newest version (3.3-4).
libncurses-dev is already the newest version (6.2-0ubuntu2).
libpcre2-dev is already the newest version (10.34-7).
libreadline-dev is already the newest version (8.0-4).
libsm-dev is already the newest version (2:1.2.3-1).
libxslt1-dev is already the newest version (1.1.34-4).
make is already the newest version (4.2.1-1.2).
patch is already the newest version (2.7.6-6).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-dev is already the newest version (3.8.2-0ubuntu2).
clang is already the newest version (1:10.0-50~exp1).
csh is already the newest version (20110502-5).
graphviz is already the newest version (2.42.2-3build2).
help2man is already the newest version (1.47.13).
libeigen3-dev is already the newest version (3.3.7-2).
libjpeg9-dev is already the newest version (1:9d-1).
libspdlog-dev is already the newest version (1:1.5.0-1).
neovim is already the newest version (0.4.3-3).
ninja-build is already the newest version (1.10.0-1build1).
swig is already the newest version (4.0.1-5build1).
tcl-dev is already the newest version (8.6.9+1).
tcllib is already the newest version (1.20+dfsg-1).
tclx8.4-dev is already the newest version (8.4.1-4).
texinfo is already the newest version (6.7.0.dfsg.2-5).
tk-dev is already the newest version (8.6.9+1).
xdot is already the newest version (1.1-2).
curl is already the newest version (7.68.0-1ubuntu2.12).
gdb is already the newest version (9.2-0ubuntu1~20.04.1).
git is already the newest version (1:2.25.1-1ubuntu3.5).
libgomp1 is already the newest version (10.3.0-1ubuntu1~20.04).
libx11-dev is already the newest version (2:1.6.9-2ubuntu1.2).
libxml2-dev is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.4).
python3-tk is already the newest version (3.8.10-0ubuntu1~20.04).
strace is already the newest version (5.5-3ubuntu1).
wget is already the newest version (1.20.3-1ubuntu2).
zlib1g is already the newest version (1:1.2.11.dfsg-2ubuntu1.3).
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
xvfb is already the newest version (2:1.20.13-1ubuntu1~20.04.3).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
To re-run with the same options:
OS=ubuntu-20.04 INSTALL_PACKAGES=yes python3 /home/monib/OpenLane/dependencies/installer.py
Copying files...
Building Python virtual environment...
Collecting pip
  Downloading pip-22.2.2-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 2.3 MB/s
Collecting wheel
  Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Installing collected packages: pip, wheel
  Attempting uninstall: pip
    Found existing installation: pip 20.0.2
    Uninstalling pip-20.0.2:
      Successfully uninstalled pip-20.0.2
Successfully installed pip-22.2.2 wheel-0.37.1
Collecting cmake
  Downloading cmake-3.24.1-py2.py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (23.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 23.3/23.3 MB 49.4 MB/s eta 0:00:00
Collecting pyinstaller
  Downloading pyinstaller-5.3-py3-none-manylinux2014_x86_64.whl (555 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 555.5/555.5 kB 77.4 MB/s eta 0:00:00
Collecting altgraph
  Downloading altgraph-0.17.2-py2.py3-none-any.whl (21 kB)
Collecting pyinstaller-hooks-contrib>=2021.4
  Downloading pyinstaller_hooks_contrib-2022.8-py2.py3-none-any.whl (239 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 239.8/239.8 kB 72.7 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./venv/lib/python3.8/site-packages (from pyinstaller->-r ../dependencies/python/compile_time.txt (line 2)) (44.0.0)
Installing collected packages: cmake, altgraph, pyinstaller-hooks-contrib, pyinstaller
Successfully installed altgraph-0.17.2 cmake-3.24.1 pyinstaller-5.3 pyinstaller-hooks-contrib-2022.8
Collecting click<9,>=8.0.0
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 2.7 MB/s eta 0:00:00
Collecting pyyaml~=5.4.0
  Downloading PyYAML-5.4.1-cp38-cp38-manylinux1_x86_64.whl (662 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 662.4/662.4 kB 8.9 MB/s eta 0:00:00
Collecting install~=1.3.5
  Downloading install-1.3.5-py3-none-any.whl (3.2 kB)
Collecting XlsxWriter~=3.0.2
  Downloading XlsxWriter-3.0.3-py3-none-any.whl (149 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 150.0/150.0 kB 38.3 MB/s eta 0:00:00
Installing collected packages: XlsxWriter, pyyaml, install, click
Successfully installed XlsxWriter-3.0.3 click-8.1.3 install-1.3.5 pyyaml-5.4.1
Installing dependencies...
[<Tool cvc (using d-m-bailey/cvc@d172016a791af3089b28070d80ad92bdfef9c585)>, <Tool magic (using rtimothyedwards/magic@f7df5e7c86fb47c5fd445c846afddc6fbabad6ae)>, <Tool netgen (using rtimothyedwards/netgen@bfb01e032f668c09ff43e889f35d611ef0e4a317)>, <Tool padring (using donn/padring@b2a64abcc8561d758c0bcb3945117dcb13bd9dca)>, <Tool vlogtoverilog (using RTimothyEdwards/qflow@a550469b63e910ede6e3022e2886bca96462c540)>, <Tool yosys (using YosysHQ/yosys@cfe940a98b08f1a5d08fb44427db155ba1f18b62)>, <Tool klayout (using KLayout/klayout@428d0fe8c941faece4eceebc54170cc04d916c03)>, <Tool openroad_app (using The-OpenROAD-Project/OpenROAD@4a99e88667b0840531ca0096c4fa0da8f80d6cb1)>, <Tool git (using git/git@e9d7761bb94f20acc98824275e317fa82436c25d)>, <Tool open_pdks (using efabless/open_pdks@44a43c23c81b45b8e774ae7a84899a5a778b6b0b)>]
Traceback (most recent call last):
  File "./env.py", line 263, in <module>
    main()
  File "./env.py", line 259, in main
    commands[args[0]]()
  File "./env.py", line 51, in local_install
    installer.run()
  File "/home/monib/OpenLane/dependencies/installer.py", line 506, in run
    install()
  File "/home/monib/OpenLane/dependencies/installer.py", line 44, in __exit__
    raise exc_value
  File "/home/monib/OpenLane/dependencies/installer.py", line 506, in run
    install()
  File "/home/monib/OpenLane/dependencies/installer.py", line 485, in install
    tool = pop()
  File "/home/monib/OpenLane/dependencies/installer.py", line 44, in __exit__
    raise exc_value
  File "/home/monib/OpenLane/dependencies/installer.py", line 438, in install
    if len(tool.dependencies):
AttributeError: 'Tool' object has no attribute 'dependencies'

donn, 2022-08-22:
Thanks for the report- but just as a heads up, I'll be deprecating the local installer very soon.

monibahmed, 2022-08-22:
Hi, will there be another way? or step to do a local install? I would like to customize some of the local tool installations (for example OpenROAD GPU enable)

donn, 2022-08-22:
There are some near-future plans to include the entire flow as just binaries you can download, but a local builder and installer has become a nightmare to maintain, as well as all the "special behavior" code to support that.

monibahmed, 2022-08-22:
Please let me know how I can help validate the flow

PR #1283 (2022-08-24): Build Script Fixes
~ Update API endpoint for `test_manifest_exists` to use v2 of the Docker API
~ Comment out `tool.dependencies` behavior in `dependencies/installer.py`
~ Add deprecation notice for the local installer

---
Resolves #1276
Resolves #1280