The-OpenROAD-Project/OpenLane #1281: couldn't create error file for command: permission denied

pengpeng-lian, 2022-08-22:
#####################error#####################################
OpenLane N/A
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

couldn't create error file for command: permission denied
    while executing
"exec tput setaf $color"
    (procedure "color_text" line 3)
    invoked from within
"color_text 6 "$message""
    (procedure "puts_info" line 3)
    invoked from within
"puts_info "Using design configuration at $::env(DESIGN_CONFIG)""
    (procedure "prep" line 100)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        if { [info exists arg_values(-file)] } {
                run_file [file normalize $a..."
    (file "/openlane/flow.tcl" line 434)

vijayank88, 2022-08-22:
@pengpeng-lian 
missing `issue_template`
Read instruction clearly and create issue as per template

pengpeng-lian, 2022-08-22:
######Instruction###########################
`./flow.tcl -design spm`

#######File permissions######################
jplian@rivai-u20-vm-intern:`~/projects/OpenLane/designs/spm/runs`$ cd ..
`drwxr-xr-x` 13 jplian intern 4096 Aug 22 16:58 runs

#######error#############################
OpenLane Container (ab6b79b):/openlane$ `./flow.tcl -design spm`
OpenLane N/A
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

couldn't create error file for command: permission denied
    while executing
"exec python3 -c "import os; print(os.path.relpath('$to', '$from'), end='')""
    (procedure "relpath" line 4)
    invoked from within
"relpath . $::env(DESIGN_CONFIG)"
    (procedure "prep" line 114)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 390)

vijayank88, 2022-08-22:
### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If neither command succeeds, you are using an out of date version of OpenLane and should probably update. 
```

pengpeng-lian, 2022-08-22:
> @pengpeng-lian missing `issue_template` Read instruction clearly and create issue as per template.
> 
> `permission denied` related to local directory access issue. Check whether you've full RW access to it or not?

@vijayank88

pengpeng-lian, 2022-08-22:
> ### Environment
> ```
> Please run the following set of commands in the OpenLane folder:
>
> make survey || python3 ./env.py issue-survey
>
> And copy and paste the ENTIRE output between the triple-backticks. [...]

I'm sorry, I'm not very good at understanding what to do after I perform that step and then what to do
##############instruction#############################
`make survey || python3 ./env.py issue-survey`
##############output########################
python3 ./env.py issue-survey
Kernel: Linux v5.4.0-122-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v19.3.15 (OK)
OpenLane Git Version: ab6b79bce7f928a9cec34ba672478398a83639e9
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

ab6b79b 2022-07-18T12:03:28+02:00 Don't require `click` for Docker platform extraction (#1205) - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
c38101e 2022-07-13T16:43:23+03:00 JSON Config Overhaul (#1134) - Mohamed Gaber -  (tag: 2022.07.14_01.36.48)
0d5084b 2022-07-13T16:42:31+03:00 Reduce resizer margins at global routing (#1168) - Anton Blanchard -  ()

vijayank88, 2022-08-22:
@pengpeng-lian 
Try following:
```
cd ~/projects/OpenLane/
make mount
./flow.tcl -design spm
```
If facing again issue copy & past all logs here

pengpeng-lian, 2022-08-22:
> @pengpeng-lian Try following:
> 
> ```
> cd ~/projects/OpenLane/
> make mount
> ./flow.tcl -design spm
> ```
> 
> If facing again issue copy & past all logs here

Still the same problem, it doesn't generate 
`runs/logs`
`runs/reports`
`runs/error.logs`
`runs/tmp`

#######instrustion##########################
`make mount`
`./flow.tcl -design spm`

#####error#########################
OpenLane N/A
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

couldn't create error file for command: permission denied
    while executing
"exec python3 -c "import os; print(os.path.relpath('$to', '$from'), end='')""
    (procedure "relpath" line 4)
    invoked from within
"relpath . $::env(DESIGN_CONFIG)"
    (procedure "prep" line 114)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 390)

vijayank88, 2022-08-22:
`python3 --version` ??

pengpeng-lian, 2022-08-22:
@vijayank88 
`Python 3.6.8`

pengpeng-lian, 2022-08-22:
@vijayank88 
jplian@rivai-u20-vm-intern:~/projects/OpenLane$ `make mount`
`cd /work/home/jplian/projects/OpenLane && \
        docker run --rm -v /work/home/jplian/projects/OpenLane:/openlane -v /work/home/jplian/projects/OpenLane/designs:/openlane/install -v /work/home/jplian/projects/OpenLane/pdks:/work/home/jplian/projects/OpenLane/pdks -e PDK_ROOT=/work/home/jplian/projects/OpenLane/pdks -e PDK=sky130A -e STD_CELL_LIBRARY=sky130_fd_sc_hd  --user 2010:3001 -e DISPLAY=localhost:16.0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /work/home/jplian/.Xauthority:/.Xauthority --network host -ti efabless/openlane:ab6b79bce7f928a9cec34ba672478398a83639e9-amd64`
OpenLane Container (ab6b79b):/openlane$` ./flow.tcl -design spm`

vijayank88, 2022-08-22:
@donn 
Can you please look into this? I'm not able to replicate this issue

donn, 2022-08-22:
No, I'm not good at unformatted chaos.

pengpeng-lian, 2022-08-22:
I'm sorry, I hope you can help me answer this time. Next time, I'll write it carefully according to the format！

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | Mohamed ***@***.***> |
| Date | 08/22/2022 22:43 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [The-OpenROAD-Project/OpenLane] couldn't create error file for command: permission denied (Issue #1281) |

No, I'm not good at unformatted chaos.

pengpeng-lian, 2022-08-22:
Thank you. I have left the company now. I may not reply to you until tomorrow. I hope you understand

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | Mohamed ***@***.***> |
| Date | 08/22/2022 22:56 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [The-OpenROAD-Project/OpenLane] couldn't create error file for command: permission denied (Issue #1281) |

Fine, I reconstructed the data. Just be more careful in the future. Also, always wrap any terminal output like this:

```
EXAMPLE
```

which shows up like this:

EXAMPLE

From what I see, you're working on some kind of virtual machine at your workplace. I do not know how your Docker environment is set up. I would wager no Docker volume mounts work at all. This could be for any number of reasons:

SELinux
A rootless docker misidentifying as a root docker setup or vice versa

Can you report the output of sestatus?

donn, 2023-04-13:
Alright, I think about a year is enough. Staled out.

closed 2023-04-13 by donn