The-OpenROAD-Project/OpenLane #1282: Magic DRC check failed

vijayank88, 2022-08-23:
### Description
With OpenLane tag [2022.08.16](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.08.16) `vsdbabysoc_wrapper` design passed Magic DRC stage without any errors.
With OpenLane tag [2022.08.17](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.08.17) flow failed with Magic DRC errors.

### Environment
```
Please run the following set of commands in the OpenLane folder:

make survey || python3 ./env.py issue-survey
python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 12a7e9ccd497950b07bf5aa1c19e34655148d60a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

12a7e9c 2022-08-17T14:09:28+05:30 [BOT] Update OpenLane tag. (#72) - github-actions[bot] -  (HEAD -> main, origin/main, origin/HEAD)
3feaa36 2022-08-15T10:17:07-03:00 Unstable s01 (#71) - vijayan -  ()
42d70a1 2022-08-11T12:49:59+05:30 [BOT] Update OpenLane tag. (#70) - github-actions[bot] -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (push)
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* Due to file size limit uploaded here: https://slack-files.com/T01699QAZBQ-F03TYEQENF8-e1eda559ee
* List the commands used to run the design.

respective CI logs can be found here:
2022.08.16 tag: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/PR-72/1/pipeline/443
2022.08.17 tag: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/PR-74/1/pipeline/497

### Expected behavior
Clean Magic drc report.

### Logs
```
[STEP 34]
[INFO]: Running Magic DRC (log: logs/signoff/34-drc.log)...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 1592
[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Skipping Tap/Decap Insertion.

    while executing
"flow_fail"

   (procedure "quit_on_magic_drc" line 13)
    invoked from within
"quit_on_magic_drc -log $::env(drc_prefix).tr"
    (procedure "run_magic_drc" line 27)
    invoked from within
"run_magic_drc"
    (procedure "run_drc_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
   invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
make: *** [Makefile:170: quick_run] Error 1
make: Leaving directory '/tmp/workspace/_Public_production-small-3_PR-74@2/OpenLane'

```
Magic DRC Screenshot
[image]
[image]

vijayank88, 2022-08-24:
@maliberty 
Reproducer attached for Magic DRC check only here.
Let me know which stage test case you're looking for, I will attach or it can be easily reproduced at your from MPW CI repo.

maliberty, 2022-08-24:
is it in the power grid?  In that case pdn, otherwise detailed routing

vijayank88, 2022-08-24:
PDN stage test case: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9415349/issue_reproducible.zip)

DRT test case: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9415354/issue_reproducible.zip)

If need macro LEF file for reference it can be found here: https://github.com/manili/vsdbabysoc_mpw3/tree/main/lef

vijayank88, 2022-08-27:
@maliberty 
Please check this issue can be fixed from OpenROAD or move to Magic issue?

maliberty, 2022-08-28:
The pipeline is now green.  Does this still need attention?

vijayank88, 2022-08-30:
@maliberty 
Sorry, I forgot to update that. I used `set ::env(QUIT_ON_MAGIC_DRC) 0`. So even though magic reports violations flow will complete.
But originally flow reports Magic DRC errors and still need to be fixed.

maliberty, 2022-09-12:
I opened the out.def in the pdn reproducer and I don't see anything that looks like the magic picture:
[image]

there is no cut shape where there is a violation shown.

donn, 2022-09-19:
@vijayank88 What's actionable here?

vijayank88, 2022-09-19:
Just checked with Latest OpenLane tag: [2022.09.16](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.16) its passed Magic DRC check without any errors.

closed 2022-09-19 by vijayank88