The-OpenROAD-Project/OpenLane #1285: LEC fails with spm design example

mattvenn, 2022-08-25:
### Description

LEC fails with spm design

### Environment
```
python3 ./env.py issue-survey
Kernel: Linux v5.13.0-28-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

### Reproduction Material

Add this line to config.tcl: set ::env(LEC_ENABLE) {1}
Then:
* make mount
* ./flow -design spm

No reproducible, openlane crashed too hard:

### Expected behavior

LEC should pass on all example designs.

### Logs
```
[INFO]: Running LEC: /openlane/designs/spm/runs/RUN_2022.08.25_15.20.58/results/placement/spm.resized.v.without_power_pins.v Vs. /openlane/designs/spm/runs/RUN_2022.08.25_15.20.58/results/synthesis/spm.v
[ERROR]: /openlane/designs/spm/runs/RUN_2022.08.25_15.20.58/results/placement/spm.resized.v.without_power_pins.v is not logically equivalent to /openlane/designs/spm/runs/RUN_2022.08.25_15.20.58/results/synthesis/spm.v.

    while executing
"logic_equiv_check -rhs $::env(PREV_NETLIST) -lhs $::env(CURRENT_NETLIST)"
    (procedure "run_resizer_design" line 19)
    invoked from within
"run_resizer_design"
    (procedure "run_placement" line 23)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 57)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 389)

```

EmJunaid, 2022-09-06:
LEC is failing with all designs.

zappos23, 2022-09-30:
Hi @EmJunaid,

I tested your LEC script fix, it works on design that do not have flop. But it fails on design that have flop. 

Thanks

kareefardi, 2023-02-02:
This should be fixed. However some designs might still fail see https://github.com/The-OpenROAD-Project/OpenLane/issues/1645

closed 2023-02-02 by kareefardi